thonny/thonny #2718: RP2040: Can not install any package/library

Terstegge, 2023-03-24:
Hi, I am using Thonny 4.0.2 with Linux Mint and a RP Pico (RP2040). Everything works nice,
but when I try to install a package (e.g. a ssd1306 driver for OLED Display, I always get a
error window saying:

install micropython-ssd1306
Error 'type' object is not subscriptable

I get this error message for any library I try to install. I am quite new to thonny and python, so
I might have overlooked something very basic...

Thanks for your help and BR

buraksahin02, 2023-03-26:
yea I took the same error. Could you solve it

aivarannamaa, 2023-03-26:
Thank you for the report!

Unfortunately I can't reproduce this with Thonny 4.0.2 on Ubuntu 22.04.

How did you install Thonny? With `pip`, https://thonny.org/installer-for-linux or some 3rd party solution?

Terstegge, 2023-03-28:
I installed Thonny with thonny-4.0.2.bash from the Thonny Website.
I just let the install script run, and now I can download packages. But I also upgrades to the
latest micropython version on my board (I compiled the latetest main version on my machine).

Anyway - the problem disappeared for me ....

TomBirchHansen, 2023-05-07:
I am having the same issue on Windows installed Thonny

closed 2023-05-07 by aivarannamaa

shreyash50111, 2024-04-01:
I also tried reinstalling, but nothing changed.

poon64, 2024-05-09:
I had the same issue both with Ubuntu, RHEL8 and Windows 11, but finally I found a solution. 
Change the interpreter to Local Python 3 and restart Thonny. 
The install the package you want, and restart Thonny after installation and change to the interpreter that you want to have. 
This worked for me, hope you get the same thing.