thonny/thonny #2720: Erro in save file python.

Sonecav2, 2023-03-24:
When I am in the process of saving the file, this error message appears.
"This file is already open in Thonny
 If you want to save with this name,
 close the existing editor first!".
The file has been saved.

CapeCrusader321, 2023-03-25:
When opening issues please try to provide details regarding source of download, system info etc.

Sonecav2, 2023-03-25:
> When opening issues please try to provide details regarding source of download, system info etc.

I edited this file and show this mensage error.

aivarannamaa, 2023-03-26:
> "This file is already open in Thonny If you want to save with this name, close the existing editor first!".

@Sonecav2, did you investigate your situation according to this message? Did you have the target file already open in another editor?

Sonecav2, 2023-03-26:
> > "This file is already open in Thonny If you want to save with this name, close the existing editor first!".
> 
> @Sonecav2, did you investigate your situation according to this message? Did you have the target file already open in another editor?

It was already open in the editor and there was no "Save" option, just "Save as" and this exception message appears, but it saves the file anyway.

aivarannamaa, 2023-03-26:
> there was no "Save" option

Do you mean the "Save" action was disabled / grayed out? This should mean that the editor was already saved.

It could be helpful if you posted the screenshot, which shows the error message and the editors.

Sonecav2, 2023-03-27:
> > there was no "Save" option
> 
> Do you mean the "Save" action was disabled / grayed out? This should mean that the editor was already saved.
> 
> It could be helpful if you posted the screenshot, which shows the error message and the editors.

as I already said that it saved the file, I can't show it here on github.

aivarannamaa, 2023-03-27:
In this case I'm afraid I don't understand the problem.

Bubbajoe57, 2023-11-07:
I'm having this same issue.  Did anyone figure out a solution? 
It says "This file is already open in Thonny. If you want to save with this name, close the existing editor first."    The "save" button is greyed out, and the message occurs when I click "Save as".  If it helps, I'm trying to save lcd_api.py to my pico from [this tutorial](https://www.youtube.com/watch?v=B8Kr_3xHjqE)  
[image]

aivarannamaa, 2023-11-18:
@Bubbajoe57, could you give me exact steps for reproducing this problem? Does it require MicroPython mode or can it be reproduced with Python 3 interpreter as well?