thonny/thonny #2722: SSL CERTIFICATE_VERIFY_FAILED on Windows

arapoport, 2023-03-27:
Hi,

I just installed Thonny under Windows and tried to install MicroPython on raspi zero. I get an SSL error (below). My Windows has all the updates, all other programs don't have any problems with SSL or access to the domains listed in the error message. 

```
Could not download variants info from https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json
Downloading from https://micropython.org/resources/firmware/rp2-pico-20220618-v1.19.1.uf2
...
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:997)>
```

I saw multiple closed tickets, but problem still exists.

aivarannamaa, 2023-03-27:
Please see #1986. The problem is that for some reason some Windows installations are not able to update some certs.

closed 2023-03-27 by aivarannamaa

arapoport, 2023-03-27:
Hi @aivarannamaa, thanks for the prompt response. 

It helped me to delete "R3" certificate from Intermediate certificates:
- Run `certmgr.msc`
- Open `Intermediate Certification Authorities\Certificates`
- Remove `R3` (it should have expired validity)

Are you sure the problem lies in Windows and not in Python? I tested ssl-connection with PowerShell and was able to make a request to the same domain using SSL (I assume PowerShell uses the Windows CAs)

The problem is that Python uses Windows certificate store and when requests go to websites that use Letsencrypt certificates then an expired R3 certificate (DST Root X3) is being applied. It's not clear if it is a problem in Windows or maybe the old Letsencrypt root certificate is buggy or maybe Python has a problem by incorrectly using Windows Certificate Store.

Regardless where the problem lies such dependency is a big obstacle for using Thonny. I followed the official Raspi Tutorial for Raspi Pico and it didn't work. Judging by number of questions on the internet it's a common problem.

References:
https://bugs.python.org/issue45372
https://community.letsencrypt.org/t/potential-problem-with-r3-intermediates-on-windows-servers/157164/14