thonny/thonny #2727: thonny 4.0.2 not support RTL language in linux

goodosuser, 2023-04-01:
salam

Unfortunately thonny 4.0.2 not support rtl language in linux, too
please solve this bug

[https://freeimage.host/i/HOKUjYg](url)

aivarannamaa, 2023-04-01:
Sorry, I can't tell what's wrong on the screenshot, but it really doesn't matter -- the text display is implemented in Thonny's UI toolkit (Tk) and if it doesn't act right, then all I can do, is wait for the next version.

closed 2023-04-01 by aivarannamaa

goodosuser, 2023-04-01:
thanks for reply

perhaps you not been see screenshot
https://freeimage.host/i/HOKUjYg

when type Persian or Arabic word like this
سلام
this word is typed م ا ل س

this bug was reported in version 3.X and was replied  : on version 4 will solved !