thonny/thonny #2728: Cannot install MicroPython or CircuitPython on rpi pico W

Nicrom098195, 2023-04-04:
Hello all!
I have a problem with my Raspberry Pi Pico W: when I install Micropython or Circuitpython, it doesn't work, and after a few seconds it resets, without having installed any firmware.
I had a similar problem with the same Pico W (#2589), but I resolved it installing CircuitPython and not MicroPython.
Now nothing works.
It worked for some minutes after a day with this problem, but when I connected a servo motor, it stopped working again.

https://user-images.githubusercontent.com/88785297/229856316-6bb48d13-99fd-43df-941e-994c12b7b285.mp4

aivarannamaa, 2023-04-05:
I suspect that this issue is not specific to Thonny. In order to be sure, please try whether the port appears after you've installed MicroPython by traditional means:

* Put the device into bootloader mode
* copy the uf2 file onto the your devices volume
* wait until the volume disappears
* check whether `/dev/ttyACM0` or `/dev/ttyUSB0` appears

If not, then try restarting your MicroPython device and/or your Linux. If it doesn't help then seek for help in a more general forum.

Nicrom098195, 2023-04-06:
It doesn't even go like that.
I will look for another forum, thanks anyway - for the help!

closed 2023-04-06 by Nicrom098195