thonny/thonny #2729: Localisation german of menu bar items missing in UI mode 'simple' in MacOS

martin-guddat, 2023-04-05:
Expected: Menu bar items are localised correctly in all UI modes

IS: In UI-mode 'regular' the Menu Bar items for `File`, `Edit`, `Window`, `Help` ... are correctly localised to german `Datei`, `Bearbeiten`. 
After switching UI mode to `simple` only the english Menu Bar item name are shown

MacOS 13.2.1 (22D68)
Thonny 4.0.2

aivarannamaa, 2023-04-20:
Thank you for the report and sorry for the late reply!

In simple mode Thonny is supposed to hide the menubar but I had forgotten that on macOS this is not possible, so that default Python menu appeared instead of Thonny's. Next Thonny version will show full (and translated) Thonny menu even in simple mode.

closed 2023-04-20 by aivarannamaa

aivarannamaa, 2023-04-25:
The first beta of 4.1.0 is available: https://github.com/thonny/thonny/releases/tag/v4.1.0b1