thonny/thonny #2730: Can not run Thonny with ESP32C3 with curcuit python

gbakerLV, 2023-04-05:
Adafruit CircuitPython 8.0.5 on 2023-03-31; Seeed Studio XIAO ESP32C3 with ESP32-C3FN4

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialTimeoutException: Write timeout).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.

aivarannamaa, 2023-04-05:
Can get CircuitPython REPL with Putty or another simpler serial program?

gbakerLV, 2023-04-10:
Yes I can.

aivarannamaa, 2023-05-25:
I now received the same device and I could reproduce this in Windows (but not always). I could not reproduce this in macOS.

aivarannamaa, 2023-05-25:
I could not reproduce the problem after I upgraded CircuitPython to 8.1.0

aivarannamaa, 2023-07-04:
Please reopen if you still have this problem with latest CircuitPython

closed 2023-07-04 by aivarannamaa

VladTheInhaler-1, 2023-12-27:
pinging as i am experiencing this same exact issue with my XIAO ESP32-C3. 
CircuitPython Version 8.2.9

I also cant copy libraries to the board, Get this error.
'
/Users/XXXXX/Desktop/adafruit-circuitpython-bundle-8.x-mpy-20231207/lib/adafruit_ble/__init__.mpy (3409 bytes)
PROBLEM IN THONNY'S BACK-END: Exception while handling 'upload' (thonny.plugins.micropython.mp_back.ProtocolError: Could not complete raw paste).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your CircuitPython device and try again.

Process ended with exit code 1.
'

aivarannamaa, 2024-02-24:
This is duplicate of #2930. It is probably a CircuitPython's issue.