thonny/thonny #2732: BUG: Thonny snaps to a new window and freezes in MacOS when tiling window to left/right of the screen

calorab, 2023-04-05:
I am on MacOS and when I try to move Thonny and chrome to left and right of the screen, that works fine. Then if I click the separation between the 2 windows (in order to drag one window wider than the other) Thonny will immediately snap to a new desktop and become non responsive. I have to force quit the program.

Caleb

Moosems, 2023-04-08:
Can you show us the bug in action?

aivarannamaa, 2023-04-09:
@calorab, which macOS version and which Thonny version are you using?

calorab, 2023-04-09:
Hi!

Yes, I took a short video of what happened
<https://drive.google.com/file/d/1QXtETIu54tlPk_BB0Q3E3WWbwwMuXfNX/view?usp=share_link>
(google
drive link). At the end of it I try clicking around and but the window is
frozen.

I'm on MacOS Monterey 12.6.3 and Thonny version 3.3.13.

Caleb

aivarannamaa, 2023-04-13:
> Thonny version 3.3.13

@calorab, you should try newer Thonny version: https://github.com/thonny/thonny/releases/download/v4.0.2/thonny-4.0.2.pkg