thonny/thonny #2733: Copy in thonny 4.0.2 in MacBookAir not working

Walia10, 2023-04-06:
I am using python4.0.2. If I try to select content inside thonny it disappears with the cursor. if I use Ctrl+A the whole programme disappears. Ctrl+C is also not working. I can copy Text from other sources to thonny . Its very annoying can someone please help me

aivarannamaa, 2023-04-09:
Thank you for the report!

Which macOS version and Thonny version are you using? Have you changed Thonny's theme? Are you using light or dark theme in macOS?

Walia10, 2023-04-18:
mac 13.2.1 and thonny 4.0.2

aivarannamaa, 2023-04-22:
I didn't notice this before -- did you really try Ctrl+A and Ctrl+C instead of Command+A and Command+C ?

Walia10, 2023-05-02:
I guess I did because earlier it automatically stopped and thonny was
working perfect. Today when I was working on thonny and selected something
in the shell thing and went back to writing my code , I wasn’t able to
select the data if I did it would disappear

aivarannamaa, 2023-05-06:
I'm confused now. Can you give me the exact steps for reproducing this problem?