thonny/thonny #2736: Thonny import msvcrt error trying to get one char at a time without enter

johngthomas, 2023-04-10:
I'm a Python raw beginner trying to get a get set number of characters typed on the keyboard recorded and displayed on screen without using the enter key. I'm on a Windows 10 desktop. Here are some of the routines I've tried in Thonny.  Am I doing something wrong here? Any suggestions?

First Attempt
import msvcrt
input_str = ""
while len(input_str) < 6:
    if msvcrt.kbhit():  # checks if a key has been pressed
        char = msvcrt.getch().decode('utf-8')  # gets the character and decodes it from bytes to string
        input_str += char
        print(char, end='')  # prints the character on the same line

Error Message:
KeyboardInterrupt: Execution interrupted
cp_back.py, line 133
No specific suggestions for this error (yet).
[image]

Second Attempt
import msvcrt
# Prompt the user to enter six characters
print("Please enter six characters:")
# Read input from the keyboard
input_chars = []
while len(input_chars) < 6:
    if msvcrt.kbhit():
        input_chars.append(msvcrt.getch().decode('utf-8'))
# Concatenate the input characters and display on one line
input_string = ''.join(input_chars)
print("You entered:", input_string)

Error Message
KeyboardInterrupt: Execution interrupted
cp_back.py, line 133
No specific suggestions for this error (yet).
[image]

Third Attmpt 
import msvcrt
digits = [] #trying digits this time
while len(digits) < 6:
    digit = msvcrt.getch()
    if digit.isdigit():
        digits.append(digit)
        print(digit.decode(), end='')

Error Message
KeyboardInterrupt: Execution interrupted
cp_back.py, line 133
No specific suggestions for this error (yet).
[image]

aivarannamaa, 2023-04-10:
You can't do this in Thonny's Shell, as it is not a proper terminal emulator. For this kind of program you need to use "Run => Run current script in Terminal".

johngthomas, 2023-04-14:
> You can't do this in Thonny's Shell, as it is not a proper terminal emulator. For this kind of program you need to use "Run => Run current script in Terminal".

Thanks. What IDE would be 'a proper terminal emulator' suitable for a beginner like me?

johngthomas, 2023-04-15:
Thankyou

closed 2023-04-15 by johngthomas