thonny/thonny #2737: `Notebook` slow and buggy on MacOS (and reportedly Windows too)

Moosems, 2023-04-10:
As documented [here](https://stackoverflow.com/questions/72664643/tkinter-notebook-loads-tabs-slow-sometimes), [here](https://user-images.githubusercontent.com/95927277/230965998-d8fa803d-9490-435e-89d6-a83e6e1bd594.mov) and [here](https://user-images.githubusercontent.com/95927277/230968004-c4c5d47b-e79e-40e1-a114-07c0be646546.mov) by me, the tcl/tk (the language and framework behind tkinter) Notebook is quite buggy on MacOS. It slowly loads tabs, sometimes shows an empty frame, often doesn't load tabs, and all this even when focusing the widgets inside. This issue is so prevalent in fact that I, along with many others who have written large programs utilizing tabs in tcl/tk have not only considered making specialized `Notebook`'s but I have started work on one in a personal project, there are plans for on in Tukaan as seen [here](https://github.com/tukaan/tukaan/issues/87) and Akuli's Porcupine once planned and possibly used a specially created one (I have been told that it used to use one by users of it and Akuli but upon inspection, I could not find this special `Notebook`). The gist of this whole error is to bring attention to the issue of the `Notebook`'s poor performance to mind with the devs and to have you consider an effort to either:
 1. Fix the issues addressed.
Or
 2. Make a special `Notebook`fixing these issues.
It is a big ask, yes, and it may be considered by some to be a small issue, but it can be quite infuriating to deal with these issues as a MacOS user whose program deals with these issues every day. You might think that the costs of this outweigh the benefits but not only would I personally help with this but I believe there are many who would use such a widget in their own projects and many who would be grateful for such a fix.

Moosems, 2023-04-13:
Update: It turns out that [Porcupine](https://github.com/Akuli/porcupine) did use a custom `Notebook` and can be seen [here](https://github.com/Akuli/porcupine/blob/2e42acbda2670993e22309a29a9668a3cc21c15f/porcupine/tabs.py). Along with that, though it may not be the best to ever exist, Futura-Py did also create one [here](https://github.com/Futura-Py/notes/blob/main/src/tabmanager.py). I have created a repo [here](https://github.com/Moosems/bazingo) to start work on creating such a Notebook and would appreciate help with it. Let me know what you think of this prospect. Thanks.

aivarannamaa, 2023-04-20:
Thank you for the suggestion. I'll consider creating a custom Notebook in order to be able to use Aqua theme on macOS.

Moosems, 2023-04-20:
I appreciate your willingness to consider such an idea. I would like to quickly clarify that it is not exclusive to the `Aqua` theme but is an issue across MacOS (the same issue happens in the `Clam` theme and others).

Moosems, 2023-04-27:
Is it possible to make the Notebook into a package so that it can be easily used by others?

aivarannamaa, 2023-04-28:
> Is it possible to make the Notebook into a package so that it can be easily used by others?

I'm afraid I won't find time for this. It's much easier to create something for single use-case than to create a universally usable and documented widget.

closed 2024-05-30 by aivarannamaa