thonny/thonny #2739: Trouble when attempting to access "manage packages:

SloptimusGrime, 2023-04-11:
Hello!
I'm attempting to install a new package in Thonny for a school project. When I open the "manage packages" window, the window freezes and the following error message appears in the shell:

"PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_active_distributions' (ImportError: The 'pyparsing' package is required; normally this is bundled with this package so if you get this warning, consult the packager of your distribution.).
See Thonny's backend.log for more info.

Process ended with exit code 1."

I'm then unable to open the manage packed window without quitting and restarting Thonny. I'm using version 4.0.2 for Windows 10, installed straight from the Thonny website. Any ideas what this could be?

aivarannamaa, 2023-04-12:
Thank you for the report!

Do you have other Python 3.10 installations in this computer? 

Have you been able to install anything in Thonny's "Manage packages" before? What precisely?

SloptimusGrime, 2023-04-13:
[image]

aivarannamaa, 2023-04-13:
Please click on the blue `C:\Users\....\site-packages` link in the lower part of the dialog. Is there anything in the folder that gets opened?

aivarannamaa, 2023-04-13:
1. Please put Thonny into Debug mode (Tools => Options => General => Debug mode checkbox)
2. restart Thonny
3. go to "Tools => Manage packages" again. Assuming you'll get the same error
4. Close the dialog and select "Tools => Open Thonny data folder"
5. Send me your backend.log

SloptimusGrime, 2023-04-18:
[backend.log](https://github.com/thonny/thonny/files/11262451/backend.log)
Here's my backend.

aivarannamaa, 2023-04-25:
Unfortunately I still don't know how to reproduce this. You could try uninstalling once more and installing https://github.com/thonny/thonny/releases/tag/v4.1.0b1

hcientist, 2025-08-20:
** TLDR: FOR US it was because we created a venv with thonny's features, and it look like we actually can solve it by running `!pip install thonny` in Thonny's python interpreter pane**

I'm using this in a classroom on computers running linux mint (more info below). We have Thonny 4.0.1 and python 3.12.3.

I have attached thonny debug mode backend log. it seems liek the relevant error was
```
15:53:38.091 DEBUG   thonny.backend: Command: InlineCommand(id='cmd_7', local_cwd='/cs/home/stu-f/stewarmc/f25/149', name='get_active_distributions')
15:53:38.092 ERROR   thonny.backend: Exception while handling 'get_active_distributions'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 282, in _handle_normal_command
    response = handler(cmd)
               ^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/thonny/plugins/cpython_backend/cp_back.py", line 480, in _cmd_get_active_distributions
    distributions=self._get_distributions_info(),
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/thonny/plugins/cpython_backend/cp_back.py", line 613, in _get_distributions_info
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

I have attached
[no-error-on-system-python3.12.3--backend.log](https://github.com/user-attachments/files/21904567/no-error-on-system-python3.12.3--backend.log)
[no-module-pkg_resources-error-on-venv-python3.12.3--backend.log](https://github.com/user-attachments/files/21904568/no-module-pkg_resources-error-on-venv-python3.12.3--backend.log)

 

# Linux Mint Info
```
System:
  Kernel: 6.8.0-71-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.8 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Desktop System: Dell product: Precision 3460 v: N/A serial: <superuser required> Chassis:
    type: 3 serial: <superuser required>
  Mobo: Dell model: 0RCPYF v: A01 serial: <superuser required> part-nu: 0AC7
    uuid: <superuser required> UEFI: Dell v: 3.17.2 date: 06/20/2025
CPU:
  Info: 24-core (8-mt/16-st) model: 13th Gen Intel Core i9-13900 bits: 64 type: MST AMCP
    smt: enabled arch: Raptor Lake rev: 1 cache: L1: 2.1 MiB L2: 32 MiB L3: 36 MiB
  Speed (MHz): avg: 1021 high: 1400 min/max: 800/5300:5600:4200 cores: 1: 801 2: 800 3: 1091
    4: 800 5: 1369 6: 800 7: 1165 8: 800 9: 1046 10: 800 11: 1273 12: 912 13: 1400 14: 800 15: 1367
    16: 1350 17: 1400 18: 1400 19: 1400 20: 1291 21: 800 22: 984 23: 1038 24: 1317 25: 800 26: 800
    27: 881 28: 800 29: 800 30: 800 31: 800 32: 800 bogomips: 127795
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel Raptor Lake-S GT1 [UHD Graphics 770] vendor: Dell driver: i915 v: kernel
    arch: Gen-13 ports: active: DP-3,HDMI-A-2 empty: DP-1, DP-2, DP-4, HDMI-A-1, HDMI-A-3, HDMI-A-4
    bus-ID: 00:02.0 chip-ID: 8086:a780 class-ID: 0300
  Device-2: HoverCam Solo8 driver: hid-generic,snd-usb-audio,usbhid,uvcvideo type: USB rev: 2.0
    speed: 480 Mb/s lanes: 1 bus-ID: 1-9:4 chip-ID: 2894:0022 class-ID: 0300
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: modesetting
    unloaded: fbdev,vesa dri: iris gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1600x900 s-dpi: 96 s-size: 423x238mm (16.65x9.37") s-diag: 485mm (19.11")
  Monitor-1: DP-3 pos: primary model: Dell U2424H serial: <filter> res: 1600x900 hz: 60 dpi: 77
    size: 527x296mm (20.75x11.65") diag: 604mm (23.8") modes: max: 1920x1080 min: 720x400
  Monitor-2: HDMI-A-2 mapped: HDMI-2 model: Seiko Epson PJ serial: <filter> res: 1600x900 hz: 60
    dpi: 25 size: 1600x900mm (62.99x35.43") modes: max: 1920x1200 min: 720x400
  API: EGL v: 1.5 hw: drv: intel iris platforms: device: 0 drv: iris device: 1 drv: swrast gbm:
    drv: iris surfaceless: drv: iris x11: drv: iris inactive: wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: intel mesa v: 24.2.8-1ubuntu1~24.04.1 glx-v: 1.4
    direct-render: yes renderer: Mesa Intel Graphics (RPL-S) device-ID: 8086:a780
Audio:
  Device-1: Intel Alder Lake-S HD Audio vendor: Dell driver: snd_hda_intel v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:7ad0 class-ID: 0403
  Device-2: HoverCam Solo8 driver: hid-generic,snd-usb-audio,usbhid,uvcvideo type: USB rev: 2.0
    speed: 480 Mb/s lanes: 1 bus-ID: 1-9:4 chip-ID: 2894:0022 class-ID: 0300
  API: ALSA v: k6.8.0-71-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
[... 35 lines cut ...]
    class-ID: 0900
  Hub-4: 2-5:2 info: Microchip (formerly SMSC) USB7206 Smart Hub ports: 4 rev: 3.2 speed: 5 Gb/s
    lanes: 1 chip-ID: 0424:7206 class-ID: 0900
Sensors:
  System Temperatures: cpu: 36.0 C mobo: N/A
  Fan Speeds (rpm): N/A
Repos:
  Packages: pm: dpkg pkgs: 3349
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //mirror.cs.jmu.edu/pub/linuxmint/packages xia main upstream import backport
    2: deb http: //mirror.cs.jmu.edu/pub/ubuntu noble main restricted universe multiverse
[... cut ...]
    5: deb http: //mirror.cs.jmu.edu/pub/ubuntu noble-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/vscode.sources
    1: deb [arch=amd64,arm64,armhf] https: //packages.microsoft.com/repos/code stable main
Info:
  Memory: total: 32 GiB note: est. available: 31.03 GiB used: 4.38 GiB (14.1%)
  Processes: 578 Power: uptime: 2d 11h 54m states: freeze,mem,disk suspend: s2idle wakeups: 0
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.3.0 Client: Cinnamon v: 6.4.8 inxi: 3.3.34
```