thonny/thonny #2741: The shell only outputs %Run -c $EDITOR_CONTENT

ohomatt, 2023-04-13:
[image]
I used the sample code of micropython, as follows:
import network
wlan = network.WLAN(network.STA_IF)
wlan. active(True)
wlan. scan()
But I only get %Run -c $EDITOR_CONTENT
But if I use: print(wlan.scan()) I get the correct output,
I tried a lot of mcu or changed the version of micropython, and it was the same. I also tried to use the Windows computer version of Thonny, and it was the same. I am not sure if it is the reason of Thonny, please forgive me if it is my reason.

ohomatt, 2023-04-15:
It should output all the wifi signals it can currently search for，like this：

ohomatt, 2023-04-15:
It should output all the wifi signals it can currently search for，like this：
[image]
This is what I print via print().

ohomatt, 2023-04-15:
Micropython is used here, and the esp8266 is controlled through the serial port to run the code

aivarannamaa, 2023-04-16:
I still don't understand what is the problem.

closed 2023-06-02 by aivarannamaa