thonny/thonny #2746: Thonny print problem

kjm1102, 2023-04-15:
Getting an error with Thonny 3.3.6 for a program that ends
for i in range (9):
   if uart.any(): reply=uart.read(160); print('reply', reply); break
 
THONNY FAILED TO EXECUTE COMMAND get_globals

SCRIPT:
__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in <dictcomp>
TypeError: 'str' object isn't callable

kjm1102, 2023-04-15:
I went looking for the line in my program causing this, it was id=time.time(), way up at the start of the program long before the print statement at the end where the crash occurred. The lesson I guess is don't use python fn names as variable names. Until today I had never heard of the python id() function. Weird that I've been getting away with this in other IDEs all these years!

aivarannamaa, 2023-04-15:
The other alternative is to use newer Thonny version (4.0.2)

kjm1102, 2023-04-16:
Is there a way to upgrade Thonny from within Thonny or do I have uninstall it, then download & install the latest version?

aivarannamaa, 2023-04-16:
If you're installing to the same directory (or keep using the standard directory), then you need to just download newer installer and install again.