thonny/thonny #2747: Thonny IDE cant detect the board

wael-ejkhidem, 2023-04-16:
hello, 
im using esp32 with cp2102, before 3 days i was able to work with the board but today i cant, i tried the board on other laptops and it works. but in my laptop it dosent. (the port is working i use it for the mouse). 
it shows the following msg :"
Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter.

Process ended with exit code None"
thank you.

aivarannamaa, 2023-04-16:
Did you try selecting specific port?

Which OS are you using?

wael-ejkhidem, 2023-04-16:
On my laptop I just have only one port, Windows 11.

aivarannamaa, 2023-04-16:
Does this one port correspond to your ESP? What happens when you select this port?

wael-ejkhidem, 2023-04-16:
I've tried to connect arduino uno on the port and its working normally.
but when i connect ESP32 the option ports doesn't appear on device manager list.

aivarannamaa, 2023-05-25:
> but when i connect ESP32 the option ports doesn't appear on device manager list.

I think in this case the problem lies outside of Thonny.

closed 2023-05-25 by aivarannamaa

wael-ejkhidem, 2023-05-25:
ok thank you, i solved the problem. but i have a problem with thonny this time. 
whenever i create a class i cant import it.