thonny/thonny #2748: PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_globals'

vishwasganatra, 2023-04-16:
I am trying to use appwrite python library for A9G module to update data on the appwrite, getting this error everytime in run the script.

My micropython code:

```import gps
import cellular
import machine
import time
from sys import maxsize

gps.on()

cellular.gprs("XXXXX", "", "")

try:
    from appwrite.client import Client
    from appwrite.services.databases import Databases
    from appwrite.services.account import Account
    from appwrite.input_file import InputFile
    from appwrite.permission import Permission
    from appwrite.role import Role
    from appwrite.id import ID
except ImportError:
    import upip
    upip.install("appwrite")
    from appwrite.client import Client
    from appwrite.services.databases import Databases
    from appwrite.services.account import Account
    from appwrite.input_file import InputFile
    from appwrite.permission import Permission
    from appwrite.role import Role
    from appwrite.id import ID

# Helper method to print green colored output.
def p(info):
    print("\033[32;1m"+str(info)+"\033[0m")

# Read the docs at https://appwrite.io/docs to get more information
# about API keys and Project IDs
client = Client()
client.set_endpoint('XXXXXXX')
client.set_project('XXXXXXX')
client.set_key('XXXXXXX')
client.set_self_signed()

databases = Databases(client)
storage = Storage(client)

database_id = "XXXXX"
collection_id = "XXXXXX"
document_id = "XXXXXXX"

def update_doc(locationString):
    p("Running updated Document API")
    response = databases.update_document(
        database_id,
        collection_id,
        document_id,
        data={
            'location-data': locationString
        },
        permissions=[
            Permission.read(Role.users()),
            Permission.update(Role.users()),
            Permission.delete(Role.users()),
        ]
    )
    document_id = response['$id']
    print(response)

while True:
    print("Location", gps.get_location())
    print("Satellites (tracked, visible)", gps.get_satellites())
    update_doc(gps.get_location())
    time.sleep(10)
```

Shell Error:

```Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "/lib/appwrite/client.py", line 67
SyntaxError: invalid syntax
>>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_globals' (thonny.plugins.micropython.mp_back.ManagementError: Script produced errors).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

aivarannamaa, 2023-04-16:
Which OS are you using? Which Thonny version?

When you experience this error again, the please select "Tools => Open Thonny data folder" and send me the backend.log from that folder.

vishwasganatra, 2023-04-16:
Am on the latest thonny version, my recent backend logs for the above script failings are:
```17:33:28.147 INFO    thonny: Thonny version: 4.0.2
17:33:28.147 INFO    thonny: cwd: C:\Users\Lenovo\Desktop\thonny\thonny-4.0.2-windows-portable
17:33:28.148 INFO    thonny: original argv: ['C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\python.exe', '-u', '-B', 'C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\lib\\site-packages\\thonny\\plugins\\micropython\\bare_metal_backend.py', "{'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'GenericBareMetalMicroPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
17:33:28.148 INFO    thonny: sys.executable: C:\Users\Lenovo\Desktop\thonny\thonny-4.0.2-windows-portable\python.exe
17:33:28.148 INFO    thonny: sys.argv: ['C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\lib\\site-packages\\thonny\\plugins\\micropython\\bare_metal_backend.py', "{'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'GenericBareMetalMicroPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
17:33:28.148 INFO    thonny: sys.path: ['C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\lib\\site-packages\\thonny\\plugins\\micropython', 'C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\python310.zip', 'C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\DLLs', 'C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\lib', 'C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable', 'C:\\Users\\Lenovo\\Desktop\\thonny\\thonny-4.0.2-windows-portable\\lib\\site-packages']
17:33:28.148 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
17:33:28.148 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': 'COM3', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'GenericBareMetalMicroPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
17:33:28.160 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
17:33:28.161 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type GenericBareMetalMicroPythonBackend
17:33:28.162 INFO    thonny: TIME/MODS 0.070 before prepare (+68 modules)
17:33:28.162 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_heapq', '_lzma', '_queue', '_tkinter', 'bz2', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'msvcrt', 'pathlib', 'platform', 'posixpath', 'queue', 'serial', 'serial.serialutil', 'serial.serialwin32', 'serial.win32', 'shlex', 'shutil', 'signal', 'subprocess', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
17:33:28.162 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
17:33:28.162 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
17:33:28.233 INFO    thonny.plugins.micropython.bare_metal_backend: Poking
17:33:28.270 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'OK\r\n'
17:33:28.276 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'PYB: soft reboot\r\n'
17:33:28.840 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
17:33:28.887 INFO    thonny: TIME/MODS 0.726 got welcome 
17:33:28.888 INFO    thonny: TIME/MODS 0.001 bef preparing helpers 
17:33:28.888 INFO    thonny.plugins.micropython.mp_back: Preparing helpers
17:33:28.888 INFO    thonny: TIME/MODS 0.000 befsubcode 
17:33:28.898 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
17:33:28.918 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
17:33:28.950 INFO    thonny.plugins.micropython.bare_metal_backend: WARNING: Could not use expected raw paste, falling back to raw
17:33:29.131 INFO    thonny: TIME/MODS 0.243 affsubcode 
17:33:29.214 INFO    thonny: TIME/MODS 0.083 affforw 
[... cut ...]
thonny.plugins.micropython.mp_back.ManagementError: Script produced errors
17:34:06.759 ERROR   thonny.plugins.micropython.mp_back: ManagementError details:
SCRIPT: __thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), __thonny_helper.builtins.id(value)) for (name, value) in __thonny_helper.builtins.globals().items() if not name.startswith('__')})

STDOUT: 

STDERR: Traceback (most recent call last):

  File "<stdin>", line 1, in <module>

  File "<stdin>", line 1, in <dictcomp>

NameError: name '__thonny_helper' isn't defined
```

vishwasganatra, 2023-04-16:
@aivarannamaa I also tried to use urequests to make network call to telegram bot seems it isn't working failing to make network call

My code:
```import gps
import cellular
import machine
import time
import urequests

gps.on()

cellular.gprs("bsnlnet", "", "")

def send_msg(text):
    token="XXXXXX:XXXXX"
    username="XXXXXX"
    url_req = "https://api.telegram.org/bot" + token + "/sendMessage" + "?chat_id=" + username + "&text=" + text
    results = urequests.get(url_req)
    print(results.json())
 

while True:
    print("Location", gps.get_location())
    print("Satellites (tracked, visible)", gps.get_satellites())
    loc = " ".join(map(str,gps.get_satellites()))
    time.sleep(3)
    send_msg(loc)
    time.sleep(7)
 ```
 
 Shell Error:
 ```Location (XXXXXXX33, XXXXXXXXXXXXXXXXXXXX7)
Satellites (tracked, visible) (0, 1)
Traceback (most recent call last):
  File "<stdin>", line 24, in <module>
  File "<stdin>", line 15, in send_msg
  File "/lib/urequests.py", line 112, in get
  File "/lib/urequests.py", line 53, in request
OSError: [Errno 9] EBADF
PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (thonny.plugins.micropython.mp_back.ManagementError: Command output was not empty).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.
```

aivarannamaa, 2023-05-25:
Sorry for the late reply!

I didn't notice it before, but your original error starts with:

```
  File "<stdin>", line 12, in <module>
  File "/lib/appwrite/client.py", line 67
SyntaxError: invalid syntax
```

Did you check this file?

The following failure of finding `_thonny_helper` may be related to https://github.com/micropython/micropython/issues/7171