thonny/thonny #2750: thonny 4.0.2: dialogs for opening/saving files don't work with zenity 3.91

befeleme, 2023-04-17:
Thonny in Fedora Linux 38 in Gnome environment, with zenity 3.91 doesn't open dialogs for opening or saving files:
Clicking on the buttons does nothing. Same when pressing the keyboard shortcuts.

This is logged when opening the file:
```
16:34:36.658 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
```
This, when saving:
```
16:35:58.465 INFO    thonny.editors: Save dialog returned None with typevariable ''
16:36:19.480 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
```
There is a known workaround (https://github.com/thonny/thonny/issues/1576#issuecomment-768346921) that works also in this case.
I decided to open the issue here as I'm not sure whether this is something to be fixed on Thonny's side or zenity's. Apparently, zenity 3.91 is still alpha (https://gitlab.gnome.org/GNOME/zenity/-/issues/54#note_1715285)

How to reproduce:
Run the Python Classroom Lab based on F38 where Thonny is preinstalled and try to open or save a file: https://labs.fedoraproject.org/prerelease 
I'll be happy to test the fix.

befeleme, 2023-04-17:
Minimally the buttons Open.. (ctrl+o), Save (ctrl+s ) and the respective options in the File menu: Open, Save, Save as, Save all files, Save copy. I have not tried everything, but New (ctrl+n), copying and pasting work correctly.

aivarannamaa, 2023-04-18:
@befeleme, @hroncok, could you please remove `--modal` and/or `--attach` parameters from the following location and try again: https://github.com/thonny/thonny/blob/be1536f930626275febc3d6a4d21327a2c0316f8/thonny/ui_utils.py#L2116 These options doesn't seem to have the expected effect anyway (at least on some systems).

If this doesn't help, then perhaps you spot another parameter in the same class which may be missing in the new zenity.

befeleme, 2023-04-18:
I did more digging in Fedora 39 with zenity 3.91:

- `--attach` results in error with the above mentioned message.
- `--modal` without `--attach` doesn't result in error (I don't know whether it's needed - without both `--modal` and `--attach` the dialog opens correctly)
-  `--name` and `--class` here:
https://github.com/thonny/thonny/blob/be1536f930626275febc3d6a4d21327a2c0316f8/thonny/ui_utils.py#L2137
- `--confirm-overwrite` when saving

Removing all of them makes it possible to open and save (as, all files, copy) files.

closed 2023-05-25 by aivarannamaa

Oussecg, 2025-05-17:
> Minimally the buttons Open.. (ctrl+o), Save (ctrl+s ) and the respective options in the File menu: Open, Save, Save as, Save all files, Save copy. I have not tried everything, but New (ctrl+n), copying and pasting work correctly.

because zenity dosen't effect new file or running the file it just open the file and save

Oussecg, 2025-05-17:
> Thonny in Fedora Linux 38 in Gnome environment, with zenity 3.91 doesn't open dialogs for opening or saving files: Clicking on the buttons does nothing. Same when pressing the keyboard shortcuts.
> 
> This is logged when opening the file:
> 
> ```
> 16:34:36.658 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
> ```
> 
> This, when saving:
> 
> ```
> 16:35:58.465 INFO    thonny.editors: Save dialog returned None with typevariable ''
> 16:36:19.480 WARNING thonny.ui_utils: Zenity returned code 255 and stderr 'This option is not available. Please see --help for all possible usages.\n'
> ```
> 
> There is a known workaround ([#1576 (comment)](https://github.com/thonny/thonny/issues/1576#issuecomment-768346921)) that works also in this case. I decided to open the issue here as I'm not sure whether this is something to be fixed on Thonny's side or zenity's. Apparently, zenity 3.91 is still alpha (https://gitlab.gnome.org/GNOME/zenity/-/issues/54#note_1715285)
> 
> How to reproduce: Run the Python Classroom Lab based on F38 where Thonny is preinstalled and try to open or save a file: https://labs.fedoraproject.org/prerelease I'll be happy to test the fix.

What version of zenity should we use ?

PR #2786 (2023-05-25): Enable thonny to open and save files with zenity 3.91+
Based on the man pages of zenity and my own testing with zenity 3.91 and 3.92 on Fedora Linux 38, this is the minimum required to open and save files in the Gnome environment. Disclaimer: this was crafted using the trial & error method and I can't really tell whether this is a proper way to fix the issue long-term. Any guidance here will be much appreciated.

- Removed `--attach`, `--name` and `--class` options which are no longer recognized.

- Removed the deprecated option `--confirm-overwrite` (man zenity 3.92.0): --confirm-overwrite DE [... cut]