thonny/thonny #2751: Formatting and Indentation Error in Thonny (works in IDLE)

Welding-Torch, 2023-04-18:
Hi Thonny Devs!

So I was writing a program that does some basic text alignment  when I realised that Thonny acts kind of weird when printing out aligned text. In the below example, I have tried to output text that is **right-aligned** in Thonny, only to find that it prints the output incorrectly. At first, I thought my code was wrong. But then I tried running it in a different interpreter(Python IDLE) on the command line and found that the output there was perfect. So I've opened this issue to bring this error in basic functionality to your attention.

I have attached the code that can be used to recreate this problem and images that showcase this problem below.

**Code:**
```python
names = ['Mark', 'Kevin', 'Paul','Leo','Knuth']
marks = [56,43,46,24,66]
for j in range(0,len(names)):
    print(f"{names[j]:>12}{marks[j]:>12}")
```

**Here's what the output is:**

[image]

Notice the improperly aligned names and numbers in this image.

**Here's what the output *should be:***

[image]

Hope this issue can be solved soon!
_Thank you for Thonny, my favourite Python IDE._

44yu5h, 2023-04-18:
Hi, What is your version of Thonny?

[image]

(btw don't hate me for using light theme)

closed 2023-04-19 by aivarannamaa

Welding-Torch, 2023-04-19:
Oh, understood. Thank you for letting me know 👍