thonny/thonny #2753: Pastes twice when CapsLock on

bgiobbe, 2023-04-19:
I'm on Thonny 4.0.2, MacOS 12.6.3 Monterey, MacBook Pro (15-inch, 2018).

When the caps lock is on, cmd-v pastes what I copied twice.
Example:
 I copy:   Y_MARGIN
 cmd-v (caps lock off) pastes:   Y_MARGIN
 cmd-v (caps lock on) pastes:   Y_MARGINY_MARGIN

aivarannamaa, 2023-04-22:
Thanks for the report!

I could reproduce this with Tk 8.6.12 but the problem seems to be fixed in Tk 8.6.13.

Most likely I'll bundle Tk 8.6.13 with the next Thonny version (4.1.0)

aivarannamaa, 2023-04-25:
First beta of 4.1.0 is now available: https://github.com/thonny/thonny/releases/tag/v4.1.0b1

closed 2023-04-25 by aivarannamaa

bgiobbe, 2023-04-26:
I installed v4.1.0b1 and tried copying and pasting with caps lock on. It
worked as expected. This bug is fixed! Thanks for tracking it down and
fixing it :)

Beth Giobbe