thonny/thonny #2754: Doesn't move selection to clipboard when cutting with capslock on

bgiobbe, 2023-04-19:
I'm on Thonny 4.0.2, MacOS 12.6.3 Monterey, MacBook Pro (15-inch, 2018).

When the caps lock is on, cmd-x deletes the selection, but seems not to put it on the clipboard.
Example:
 * caps lock off
     * Select Y_MARGIN
     * cmd-x
     * cmd-v :  pastes Y_MARGIN
 * caps lock on
     * Select Y_MARGIN
     * cmd-x 
     * cmd-v: pastes nothing

aivarannamaa, 2023-04-22:
Thanks for the report!

I could reproduce this with Tk 8.6.12 but the problem seems to be fixed in Tk 8.6.13.

Most likely I'll bundle Tk 8.6.13 with the next Thonny version (4.1.0)

closed 2023-04-23 by aivarannamaa

bgiobbe, 2023-04-26:
I installed v4.1.0b1 and tried cutting and pasting with caps lock on. It
worked as expected. This bug is fixed! Thanks for tracking it down and
fixing it :)

Beth Giobbe