thonny/thonny #2759: Shell error if freezing the program, run id grayed out, and the shell error script won't clear

elemenel, 2023-04-21:
─The shell remains the same, even after I restart Thonny. It is frozen.──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 187, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.10/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/home/elemen/.local/lib/python3.10/site-packages/thonny/plugins/cpython_backend/__init__.py", line 5, in <module>
    from .cp_back import MainCPythonBackend, get_backend
  File "/home/elemen/.local/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_back.py", line 23, in <module>
    from thonny.backend import MainBackend, logger
  File "/home/elemen/.local/lib/python3.10/site-packages/thonny/backend.py", line 5, in <module>
    import pathlib
  File "/home/elemen/.local/lib/python3.10/site-packages/pathlib.py", line 10, in <module>
    from collections import Sequence
ImportError: cannot import name 'Sequence' from 'collections' (/usr/lib/python3.10/collections/__init__.py)

Process ended with exit code 1.

elemenel, 2023-04-21:
It might have something to do with my file permissions. I will look into that, and I think this will clear up. Thanks

aivarannamaa, 2023-04-23:
Do you happen to have a file named "collections.py" in your working directory?

elemenel, 2023-04-25:
No, but here is more background into this issue, which I have resolved.  I am running Thonny on an HP EliteDesk 800 G1 SFF, with Kubuntu OS. 

I am a retired program manager, and have picked up programming Python turtle as a hobby since 2020. My preferred IDE is Thonny. I like the lack of clutter, the self-contained pip gui, and the reliabilty. 

A couple of weeks ago the motherboard on my desktop pc died, likely due to the heavy graphics load that it absorbed. After replacing the motherboard and adding a new graphics board, I attempted to install Thonny, and this is where I learned how the installation source can dictate the outcome. First, I installed Thonny 4.02 from the Ubuntu snap store. That is when I ran into the permissions issues. I then tried installing Thonny from pip, but only version 3 was available. And finally, I used the installer from the Thonny website, and it installed perfectly! That is where I should have started from. This was definitely user error, and because I am learning as I go, I stumble often. 

I appreciate your support, thanks!

closed 2023-04-25 by elemenel