thonny/thonny #2762: Cut and Past not working between Raspberry 32 Bit OS and MacOS

FlydGitHub, 2023-04-26:
Cut and Past not working between Raspberry 32 Bit OS and MacOS

Any suggestions would be welcome.

Thanks

aivarannamaa, 2023-04-26:
What do you mean by "between"? Gan you give me the steps for reproducing this problem?

FlydGitHub, 2023-04-26:
Hi. and thank you.

Sorry it's inconsistent. Sometime it works, most time not. I can't put my finger on it. 

Coping from Raspberry Pi OS to my Intel Mac.

aivarannamaa, 2023-04-26:
I don't understand. What are you trying to copy from one machine to another?

FlydGitHub, 2023-04-26:
All current versions.

What does 'Which Tk version?' ???? refer to?

FlydGitHub, 2023-04-27:
About Thonny 

Thonny 4.0.1 https://thonny.orq  
Raspbian GNU/Linux 11 (bullseye) (64-bit) Python 3_9.2 (32 bit) TIC 8.6.11 
Made at the University of Tartu, Estonia, with help from the open-source community,  Raspberry Pi Foundation  and Cybernetica AS  
Copyright (9) 2023 Aivar .Annarraa This program comes with .ABSOLLTELY NO WARRANTY! It is free software.. and you are welcome to redistribute it under certain conditions.. see httpsllopen SOU rce.orgilicensesolvilT for details 
OK

FlydGitHub, 2023-04-28:
It appears I can use the shell to cut and past reliably. But as I have said not in the min Thonny editor.

aivarannamaa, 2023-09-27:
Please see, whether the latest release (https://github.com/thonny/thonny/releases/tag/v4.1.3) works for you