thonny/thonny #2763: Missing Interpreter tab in 4.1.0b1 Mac build

rtideas, 2023-04-26:
Trying to use my ESP32-S2 with the latest beta 4.1.0b1 and using the lower right menu (or Preferences) and the Interpreter tab has disappeared. 

Accessing the menu in 4.1.0b1:
[image]

Where's the Interpreter tab?
[image]

Installed back and forth a few times between 4.1.0b1 and 4.0.2 with same issue. **Had to revert to 4.0.2 to still use my ESPs.**
[image]

aivarannamaa, 2023-04-26:
Thank you for the report!

Actually I've already noticed and fixed it (115ab502) but haven't been able to make a new release yet. I'll do it soon.

wohltat, 2023-05-29:
I can replicate the bug on linux also. It is solved in the current git version.

aivarannamaa, 2023-06-03:
Fix released now: https://github.com/thonny/thonny/releases/tag/v4.1.0

closed 2023-06-03 by aivarannamaa