thonny/thonny #2766: Using GTK3 with Weasyprint in Thonny (Portable)

TZimm55, 2023-04-27:
To save myself from rewriting everything I have outlined my problem on [stackoverflow](https://stackoverflow.com/questions/76121938/using-gtk3-with-weasyprint-in-thonny-portable).

In short, with luck I have gotten weasyprint to work after installing weasyprint and GTK3, adding GTK3 to PATH and using VSCode.

What I am trying to do is get it to work with the portable version of Thonny, so I can take it on the go.

Thank you!

aivarannamaa, 2023-04-27:
> I have installed Thonny, then moved the Thonny file to a USB to make it portable.

There's separate portable variant available: https://github.com/thonny/thonny/releases/download/v4.0.2/thonny-4.0.2-windows-portable.zip.

> I installed weasyprint via Thonny's plug-in manager.

Do you mean "Tools => Install plug-ins"? This is meant for plug-ins, not for packages used in your programs. Make sure you've read the text on top of the dialog window.

TZimm55, 2023-04-27:
@aivarannamaa thank you for responding. 

I did not realize there was a pre-packaged portable version. Do I need to switch to the portable version? My Thonny application says “Portable Thonny” at the top. 

And you are correct. I think I might have been trying the plug-in menu instead of the packages.

I am going to try and do the “install from requirements file” from the packages menu. See if that works. I am not seeing a GTK that matches the GTK GitHub page I am downloading from.

TZimm55, 2023-04-27:
I used the “Manages packages” window and did install from requirements. It looks like it took the zip file and made a new file in Thonny’s site-packages folder but named it “Unknown”. It also put a folder on my local C:\ deep in AppData. This still didn’t work.

aivarannamaa, 2023-04-28:
I'm afraid I don't have time to help here, especially as WeasyPrint documentation says "Installing WeasyPrint on Windows requires to follow a few steps that may not be easy". 

If you suspect that Thonny is doing something wrong, then send me the exact steps for reproducing the problem and description of expected and actual behavior.

closed 2023-04-28 by aivarannamaa

TZimm55, 2023-04-28:
@aivarannamaa, no worries, thank you for your time. Good job on Thonny!