thonny/thonny #2770: Cant clear the shell

VyshnavArun, 2023-05-03:
shows this error
Traceback (most recent call last):
  File "C:\Users\Vychu\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Vychu\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1262, in _clear_shell
    self._clear_content(end_index)
  File "C:\Users\Vychu\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1335, in _clear_content
    btn_pos = float(self.index(btn))
ValueError: could not convert string to float:

aivarannamaa, 2023-05-07:
Please try the newest Thonny version as this problem may be already fixed.

aivarannamaa, 2023-07-16:
Please reopen if you still have this problem

closed 2023-07-16 by aivarannamaa