thonny/thonny #2775: venv plug-in

rootsmusic, 2023-05-07:
I've installed Thonny 4.0.2 on Windows 11.  How can I install [venv](https://github.com/python/cpython/tree/3.7/Lib/venv/) to be able to create environments?  Thanks.

aivarannamaa, 2023-05-07:
The package `venv` comes together with Python on Windows and Mac. This means you should already have this package. 

Ar you experiencing any problems?

rootsmusic, 2023-05-07:
Yes @aivarannamaa, Python 3.10.9 was installed by Thonny.  In Thonny's Python Shell, I tried to use venv but it doesn't seem to be available:

`python -m venv`
>           ^^^
> SyntaxError: invalid syntax

aivarannamaa, 2023-05-07:
`python -m venv`, is supposed to be executed in the system shell (ie. via Windows Terminal or cmd.exe) not in Python Shell (a.k.a REPL).

rootsmusic, 2023-05-07:
Sorry @aivarannamaa, I didn't know.  In Command Prompt (cmd.exe), I'm able to use venv.  I've activated the new environment, and it created a Scripts directory containing the following files:
05/07/2023  10:59 AM             2,087 activate
05/07/2023  10:59 AM             1,016 activate.bat
05/07/2023  10:59 AM            24,167 Activate.ps1
05/07/2023  10:59 AM               393 deactivate.bat
05/07/2023  11:03 AM           106,373 pip.exe
05/07/2023  11:03 AM           106,373 pip3.10.exe
05/07/2023  11:03 AM           106,373 pip3.exe
05/07/2023  10:59 AM           266,624 python.exe
05/07/2023  10:59 AM           254,848 pythonw.exe

For this new environment, I want to use Python 3.7.7.  When I execute `pip uninstall python`, the error is: "Skipping python as it is not installed."  When I execute `pip install python==3.7.7`, the error is: "ERROR: Could not find a version that satisfies the requirement python==3.7.7 (from versions: none)".  Am I doing it wrong?  Thanks.

aivarannamaa, 2023-05-07:
You seem to have quite a confusion here. Please see https://realpython.com/python-virtual-environments-a-primer/

closed 2023-05-07 by rootsmusic