thonny/thonny #2776: Building from source

CapeCrusader321, 2023-05-07:
Hey @aivarannamaa  is there any way i can simply clone this repo and build thonny from scratch for linux, if its possible can you also point me to the guide or wiki showing building from this source ??

aivarannamaa, 2023-05-07:
Do you mean building the binary bundle with Python, Tk etc?

Please see the folder `packaging/linux` in this repo. What I do is run following scripts on Ubuntu 18.04:

1. install_tools_ubuntu.sh (once)
2. build_requirements.sh (each time I upgrade a dependency -- see line 7 in build_python.sh for an example)
3. install_tarball.sh (for each release)

I'm using an older distribution for building in order to not introduce dependency on newer glibc versions.

CapeCrusader321, 2023-05-08:
Somewhat similar, what i was wishing is that currently snap builds of thonny depends on tar.gz releases of thonny that comes without the desktop file, icon etc. plus i think it will lag in future for latest thonny versions, so i was searching for a way with which i can simply clone this directory and automate snap builds of thonny, will try the above mentioned although its difficult to execute above steps in snap, for example in gtk apps we simply have a meson file etc. i clone it and execute the meson file with dependencies it builds a snap package with few efforts i believe its not possible with python toolkits like tkinter etc ?

closed 2023-05-08 by CapeCrusader321