thonny/thonny #2779: installing Thonny into Python on Windows

rootsmusic, 2023-05-11:
I installed [Python 3.7.7](https://www.python.org/downloads/release/python-377/) in Windows 11 to C:\Program Files\Python37.  Then I [installed Thonny](https://github.com/thonny/thonny/wiki/Windows#installing-thonny-and-python-separately) separately to "C:\Program Files\Python37\Scripts\thonny.exe".  I didn't get any error, but I can't run thonny.exe.

aivarannamaa, 2023-05-12:
> I didn't get any error, but I can't run thonny.exe.

Can it be that the error did appear, but you couldn't see it because the terminal window was closed first? In this case open the terminal first and then launch thonny.exe

rootsmusic, 2023-05-12:
@aivarannamaa, which terminal are you referring to?  In Command Prompt, I tried to run "C:\Program Files\Python37\Scripts\thonny.exe" in the Command Prompt.  I also tried to click on Thonny's shortcuts on my Desktop and in my Start Menu.  No terminal opens, and nothing happens.

aivarannamaa, 2023-05-13:
>  In Command Prompt, I've tried to manually run "C:\Program Files\Python37\Scripts\thonny.exe"

What happened? Did you get any output?

please also check Thonny's logs in `C:\<username>\AppData\Roaming\Thonny` (can be easily opened by pasting `%APPDATA%\Thonny` into Windows Explorer's "address bar". Is there anything in frontend.log or frontend_faults.log?

rootsmusic, 2023-05-13:
@aivarannamaa No matter how I try to run Thonny, nothing is displayed.  However, the date stamps for frontend.log and frontend_faults.log update to the time that I tried to run Thonny.  The latter file is blank though, and frontend.log only shows: "INFO    thonny: Thonny version: 3.3.14".

aivarannamaa, 2023-05-15:
No idea, what can be the reason.

If you want to use Python 3.7 for running your programs, then you can still use the latest Thonny version and let it run with bundled Python 3.10, but configure it to use another interpreter for user programs (Run => Configure interpreter). You need to uninstall Thonny from Python 3.7 first, though.

rootsmusic, 2023-05-15:
@aivarannamaa I've downloaded the latest version, which is newer than version 3.3.14 that I uninstalled in this issue.