thonny/thonny #2785: esp32 webrepl not working

Shafikulachalil, 2023-05-16:
Can't import `websockets`. You can install it via 'Tools => Manage plug-ins'.

Process ended with exit code 1.

aivarannamaa, 2023-05-25:
Did you try installing websockets?

Shafikulachalil, 2023-06-01:
its worked after re flashing micropython

closed 2023-06-01 by aivarannamaa

akomakom, 2023-06-13:
For anyone finding this with the same problem, this thread may be helpful:
https://forums.raspberrypi.com/viewtopic.php?t=348865

In Ubuntu, I was able to resolve this by uninstalling the websockets plugin from thonny, then installing it via pip3 as root:
`sudo pip3 install websockets`.

There is most likely a bug in thonny:
1. Tools -> Open System Shell
2. Run
 ```
import sys
print(sys.path)
```

Note that path of websockets plugin (shown in Plug-In Manager) isn't in this list.