thonny/thonny #2789: ssh connection disconnects without jedi on target server

ScoreUnder, 2023-05-17:
When using python3 over ssh:

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'highlight_occurrences' (ModuleNotFoundError: No module named 'jedi').
See Thonny's backend.log for more info.

Connection lost -- 

Use Stop/Restart to reconnect.

Process ended with exit code 1.
```

I would expect either a fall back to unhighlighted code, or to locally highlighted code, rather than a disconnect in this circumstance.

aivarannamaa, 2023-05-17:
Thanks for the report! I'll handle this.

elschopi, 2024-07-14:
I try to use the remote python feature of Thonny 4.1.4 (on Win 11 64bit) with the target being DietPi 9.6.1 on a Raspberry Pi Zero W (switched SSH server from Dropbear to OpenSSH). I can connect using my SSH credentials, but after a few seconds, get dropped with the "no module named 'jedi'" error message. Is there a beta version with fixes implemented or do I need to install any additional libraries to the Pi?