thonny/thonny #2796: import classes

wael-ejkhidem, 2023-05-25:
hello, 
i cant import any class using thonny, the classes that i can import is the one that made by a programmer or by me. 
it says the following error message: 
"Traceback (most recent call last):
  File "<stdin>", line 29, in <module>
ImportError: no module named 'hc_sr04'"

aivarannamaa, 2023-05-25:
Are you sure the module is present in a correct location?

wael-ejkhidem, 2023-05-25:
yes, i can see the module on the right window.

aivarannamaa, 2023-05-25:
Can you post the screenshot with file-browser, main file and error output?

wael-ejkhidem, 2023-05-25:
[image]
[image]
[image]

aivarannamaa, 2023-05-27:
I see -- the code running on your ESP device is trying to import a module from your PC's file system. This is not possible. You need to upload all the required modules to your device.

It's slightly different with the main module -- when you press F5, Thonny sends the text of the main module to MicroPython REPL, ie. it does not have to be in device's filesystem. But importing always involves reading from a file.