thonny/thonny #2797: No connection to esp32 via ftdi

Jenatsch, 2023-05-25:
Flashing works perfectly but then the connection to the repl fails with following message:
Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython / firmware;
  - make sure the device is not in bootloader mode.

But if I connect via HTerm it works perfectly:
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information
>>> 

I also tried with the newest firmware version [v1.20.0] with the same result.

aivarannamaa, 2023-05-27:
Thank you for the report!

Please try this:

1. turn on debug mode (Tools => Options => General => 5th checkbox)
2. restart Thonny
3. wait until you see the same "Device is busy or does not respond" notification
4. Select "Tools => Open Thonny data folder"
5. Post the contents of backend.log

Jenatsch, 2023-05-28:
22:47:45.743 INFO    thonny: Thonny version: 4.0.2
22:47:45.743 INFO    thonny: cwd: C:\Users\jenatsch\AppData\Local\Programs\Thonny
22:47:45.744 INFO    thonny: original argv: ['C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\python.exe', '-u', '-B', 'C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM8', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:47:45.744 INFO    thonny: sys.executable: C:\Users\jenatsch\AppData\Local\Programs\Thonny\python.exe
22:47:45.744 INFO    thonny: sys.argv: ['C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM8', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:47:45.744 INFO    thonny: sys.path: ['C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp', 'C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\python310.zip', 'C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\DLLs', 'C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\lib', 'C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny', 'C:\\Users\\jenatsch\\AppData\\Local\\Programs\\Thonny\\lib\\site-packages']
22:47:45.744 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
22:47:45.746 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': 'COM8', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
22:47:45.752 DEBUG   thonny.plugins.micropython.serial_connection: Opening serial port COM8
22:47:45.770 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
22:47:45.771 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
22:47:45.773 INFO    thonny: TIME/MODS 0.290 before prepare (+73 modules)
22:47:45.774 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_heapq', '_lzma', '_queue', '_struct', '_tkinter', 'binascii', 'bz2', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'msvcrt', 'pathlib', 'platform', 'posixpath', 'queue', 'serial', 'serial.serialutil', 'serial.serialwin32', 'serial.win32', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
22:47:45.774 DEBUG   thonny.plugins.micropython.bare_metal_backend: _process_until_initial_prompt, clean=True
22:47:45.775 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:47:45.775 DEBUG   thonny.plugins.micropython.mp_back: Sending ctrl chars: b'\x03'
22:47:45.776 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
22:47:45.842 INFO    thonny.plugins.micropython.bare_metal_backend: Poking
22:47:45.842 DEBUG   thonny.plugins.micropython.mp_back: Sending ctrl chars: b'\x01'
22:47:45.906 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:47:45.906 DEBUG   thonny.plugins.micropython.mp_back: Sending ctrl chars: b'\x03'
22:47:45.906 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
22:47:46.033 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:47:46.033 DEBUG   thonny.plugins.micropython.mp_back: Sending ctrl chars: b'\x03'
22:47:46.034 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
22:47:48.007 INFO    thonny.plugins.micropython.bare_metal_backend: Giving advice

Jenatsch, 2023-05-28:
Maybe there should be something in RTS DTR, becuase these pins are connected to the esp32?

dadalee51, 2023-05-30:
@aivarannamaa  It's not easy to check what happened in the thonny backend with debug turned on, I can replicate this issue by flashing micropython firmware, reconnect USB, then thonny repl will not respond.  I even got up to a stage that the Interpreter tab disappeared. OS: LinuxMint 21 Thonny: thonny-4.1.0b1. 
Funny thing is, after reflash the firmware, everything works fine until USB gets disconnected.

Jenatsch, 2023-05-30:
Interesting, in my case i never works. The error message appears also directly after flashig it.

dadalee51, 2023-05-31:
after a bit of fiddling I noticed my hardware was a esp32 behind a CH340 controller with DTR and RTS pin connection very much like the schema in the following link: 

https://pcbartists.com/design/embedded/esp32-ch340-programmer-schematic/

and require following triggering sequence to start up:
import serial
ser = serial.Serial('/dev/ttyUSB0', baudrate=115200)
ser.setDTR(True)  # Set DTR signal to active (logical high)
ser.setRTS(True)  # Set RTS signal to active (logical high)
ser.setDTR(False)  # Set DTR signal to inactive (logical low)
ser.setRTS(False)  # Set RTS signal to inactive (logical low)
ser.close()

hope this helps.

dadalee51, 2023-05-31:
@Jenatsch 
if your esp32 hardware has a USB UART interface chip that looks like this:
[image]
I did following to get my micropython chevron back:
1. in thonny Tools->Open Thonny data folder
2. close thonny ide
3. edit configuration.ini file as follows:
4. [image]
5. start thonny again
6. micropython prompt should come back.
7. thonny version :
[image]
Hope this helps.

aivarannamaa, 2023-05-31:
@Jenatsch, thanks for this information! I'm considering turning `dtr` and `rts` off for certain devices automatically.

Please give me more details about your FTDI adapter by selecting "Thonny's Python" as interpreter and running following script:

```python
import sys
if sys.platform == "darwin":
    from adafruit_board_toolkit._list_ports_osx import comports
elif sys.platform == "win32":
    from adafruit_board_toolkit._list_ports_windows import comports
else:
    from serial.tools.list_ports import comports

for p in comports():
    for key in ["device", "name", "description", "hwid", "vid", "pid", "serial_number",
                "location", "manufacturer", "product", "interface"]:
        print(key + ":", getattr(p, key))
    print("-------------")
```

dadalee51, 2023-05-31:
@aivarannamaa 

device: /dev/ttyUSB0
name: ttyUSB0
description: USB Serial
hwid: USB VID:PID=1A86:7523 LOCATION=3-2
vid: 6790
pid: 29987
serial_number: None
location: 3-2
manufacturer: None
product: USB Serial
interface: None
-------------

Jenatsch, 2023-05-31:
@dadalee51 
Thanks a lot. Now it works!
My configuration is a bit different. I use this external FTDI Chip:
[image]
thats hooked up to the esp32 like this:
[image]

@aivarannamaa 
device: COM8
name: COM8
description: USB Serial Port (COM8)
hwid: USB VID:PID=0403:6001 SER=A50285BIA
vid: 1027
pid: 24577
serial_number: A50285BIA
location: None
manufacturer: FTDI
product: None
interface: USB Serial Port
-------------

aivarannamaa, 2023-06-02:
I'll deal with DTR and RST options under #2764