thonny/thonny #2801: Flashing micropython does not work with ESP-01

wohltat, 2023-05-29:
I cannot successfully flash micropython from the thonny gui anymore. It worked in previous versions.
It uploads without errors but when i try to run the TX or RX indication LED blinks rapidly without stopping. 
When i try to connect i just see gibberish like the following on the console:

``` 
��nn'l`�nn$`nr���nl`r��nl`��p�nn�llllb'|l`rl�l�b|���rblb�nnlnn�bbp�$blrlp�n��lbn�n��$�b��nn'l�l`�nn$`nr���n�`p�n�r�bn�nll�b��nn'l`�nn$`nr���n��l`p�n�r�bn�nlb��nn'l`�nn$`nr���nl`r��nl`��p�nn�llllb'|l`rl�l�b|
```
[image]

I manually flashed the v1.20 micropython on an ESP-01 (ESP8266) and everything works fine. No unstoppable LED blinking and i can connect to the board with thonny.
`> build/thonny/thonny_venv/bin/esptool.py --port /dev/ttyUSB* --baud 460800 write_flash --flash_size=detect 0 Downloads/esp8266-1m-20230426-v1.20.0.bin`

I had a closer look on the upload log in thonny and i copied the command line. The culprit seems to be `--flash_size keep`. If i change it to `--flash_size detect` it works again. Everything in the output looks identical except for the line `Auto-detected Flash size: 1MB`.

What about adding that to the list of `Flash mode` or even make it the default?

-----------

version: Thonny 4.1.0b1

aivarannamaa, 2023-05-30:
Thanks for the report!

As `keep` is the default in recent versions of esptools (since 3.0), I'd like to keep it as default in Thonny as well. But I'm happy to add a dropdown for `--flash-size`, just after the dropdown for `--flash-mode`.

aivarannamaa, 2023-05-30:
I see that you are using the master state of Thonny's code. I now pushed the addition of flash-size dropdown. Please see, if this works for you!

wohltat, 2023-05-30:
Just tested it. Seems to work.

It would be nice if the settings were stored, so that it doesn't have to be entered every time.
In my case that is 5 dropdown settings.

aivarannamaa, 2023-06-02:
I'll skip the stored settings for now. I'll reconsider this when more users want it.

closed 2023-06-02 by aivarannamaa

jouellnyc, 2024-06-10:
To reinforce this. For anyone who got here via Google like me. I had a similar issue. 

To summarize:

I had 5 esp01s's with 4 MB flash from AMZN. No matter what I did I **could flash** but **not boot** the esp01s's. I tried all flash options: dio, qio,dout and quot - did not work...

BUT!!! I stumbled on this post and explicitly set the flash size to 4MB vs keep. 

I also set the flash mode to QIO, which is supposed to the most common as per: https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/flash-modes.html

That resolved it.