thonny/thonny #2803: Need for udisks or mount  by thonny ??

CapeCrusader321, 2023-05-31:
Hello @aivarannamaa  i wanna ask you about some quirks of thonny,  i want to know why do thonny uses udisk and is access to udisk necessary for thonny, recently in the snap package few things of thonny were not working due to the lack of access to udisks2 plug of snapcraft, to fix it i requested the plug from snapcraft reviewers and tgey would like to know for what does thonny uses udisks2 and is it a essential need for thonny, plus i would also like to know its relation with raspberry devices etc.

aivarannamaa, 2023-05-31:
This change was introduced in #2683. @jwillikers can probably explain and offer some advice.

jwillikers, 2023-05-31:
@CapeCrusader321 Yeah, I can explain that one. Thonny supports updating the firmware of Micropython and CircuitPython microcontrollers, like the Raspberry Pi Pico. Many of these devices are updated by exposing a volume over a USB connection where the user can drop a UF2 firmware. Thonny can take care of most of this process, but requires searching through the mounted devices in order to find the appropriate microcontroller. There were several issues that showed up in Thonny's issue tracker related to this functionality, so it does appear to be a use case that users expect. I'd think that the Snap would require this functionality just as the Flatpak does.

Previously, Thonny only supported this by using the `mount` command directly. I added the Udisks functionality since it is the primary way for users to mount disks on Linux and it is available from within sandboxes like the Flatpak, which do not contain the `mount` command.

CapeCrusader321, 2023-05-31:
@jwillikers  Thanks for the explanation

CapeCrusader321, 2023-06-08:
The plug is granted, thansk @aivarannamaa  and @jwillikers  for the help

closed 2023-06-08 by CapeCrusader321