thonny/thonny #2804: Thonny always save the file when run even if there is nothing to save

nicolasdespres, 2023-05-31:
Hi,

When clicking the "run" button (or pressing F5), Thonny always save the file even if there is nothing to save.

This is a major problem when the script is inside a non-writable directory, since it prevents the script from running. This happen quite frequently in school directory layout where a folder is writable only by the teacher.

Previous version of Thonny did save the file only when it was modified.

I observed this regression with version 574203fa40f5ebeca7b4a2199f757fbecef15563.

Cheers and thanks a lot for all your work on Thonny,

aivarannamaa, 2023-06-02:
Thank you for the report! 

Are you sure you referenced the right commit? 574203fa40f5ebeca7b4a2199f757fbecef15563 deals with a different topic.

nicolasdespres, 2023-06-05:
Sorry for the late reply.

I meant that I tested with #574203f, but it was introduced earlier.

aivarannamaa, 2023-08-05:
The fix will be in 4.1.2

closed 2023-08-05 by aivarannamaa