thonny/thonny #2812: Issue with editing text within triple quotes, seems to see the tab key as asking for autocomplete instead of simply tabbing along for commenting's sake

virgosupercluster23, 2023-06-09:
[image]

In this image i have captured an example of wanting to have a bunch of (as it happens) dictionary keys
but say theyre just **triple quote encapsulated lines**, and i want to use the tab key to pad spacing across
to a common distance across lines, where i can comment on them all line by line, for notes on their
values contents, i.e. which ones in this case were interesting to my use case, empty, etc
On using the tab key like i would a text editor, many IDE's as well, it complains that there is a problem
with autocomplete, which i have no interest in using, autocomplete is taking precedence over the use
of the tab key that i am intending, not the autocomplete use of the tab case it is intending.

Its just some comment section, and i want to make notes inside that comment section, but its either
trying to evaluate it as code or otherwise fail to treat it as text as being inside triple quotes should allow.

virgosupercluster23, 2023-06-09:
I was trying to tab along on line 23 if that makes it more specific

aivarannamaa, 2023-06-09:
You seem to have an old version of Thonny. Please try the latest version ([4.1.0](https://github.com/thonny/thonny/releases/tag/v4.1.0)) and also check your settings in "Tools => Options => Editor"