thonny/thonny #2813: SQLite from Thonny

mfriedenthal, 2023-06-10:
I've been teaching Intro to Python using Thonny at a library. Two years ago I added a class on SQL, also taught from the Thonny environment.

I wrote a bunch of scripts to demonstrate concepts, beginning each one with:

`import sqlite3

connection = sqlite3.connect("C:\\Python\sm_app.sqlite")
cursor = connection.cursor()`

Followed by executing the cursor with the specific SQL script.
While this works on the computers Thonny was installed onto a few years ago, it does not work for new installations. The error message is:
sqlite3.Operation Error: unable to open database

Can someone explain the error and show me how to connect to sqlite in Thonny 4.0.2 and Python 3.10.9 64 bit on Win 11 machines?

aivarannamaa, 2023-06-11:
With `"C:\Python\sm_app.sqlite"` you're relying on `\P` and `\s` not being escape sequences for Python string literals. You wouldn't be so lucky with e.g. `"C:\Python\bm_app.sqlite"`. 

Better use `r"C:\Python\sm_app.sqlite"` or ``"C:\\Python\\sm_app.sqlite"` or ``"C:/Python/sm_app.sqlite"`. If still no success, then you need to investigate your database. Perhaps it's not compatible with newer SQLite.

Anyways, it doesn't look like Thonny-specific issue.

mfriedenthal, 2023-06-11:
I’ll try that when I’m back in my office, and thank you for the suggestion. But, as I said, this string works fine on all the installations older than a few months ago, perhaps older than February.

KubaO, 2023-11-04:
> this string works fine on all the installations older than a few months ago, perhaps older than February.

Buggy code is not guaranteed to work forever. The backslash-using had a bug and eventually it stopped working.

Alas, you should not be writing to "C:\Python" anyway. Only write to directories under user's home directory. On windows, that would be under `pathlib.path.Home()`.