thonny/thonny #2814: Thonny cannot connect to esp32, but gtkterm can

jouellnyc, 2023-06-11:
Basically I can flash the esp32 with this firmware https://github.com/shariltumin/esp32-cam-micropython-2022 using esptool and can query it:

```
esptool.py v4.6-dev
Serial port /dev/ttyUSB0
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 0c:b8:15:f5:b8:b8
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

```

Thonny:

[image]

Does not seem to connect to it, but gtkterm does:

[image]

I have been pretty diligent in my tshoots, I have a few boards and most are failing due to hardware or some other issue. But seeing gtmterm can connect, it seems like Thonny should, so I am reporting.

Thonny shows:

[image]

I don't see much in the way of log files generated...

jouellnyc, 2023-06-26:
Maybe these logs help:

```
cat backend.log 

22:17:37.766 INFO    thonny: Thonny version: 4.0.2
22:17:37.767 INFO    thonny: cwd: /home/myhome/scripts
22:17:37.767 INFO    thonny: original argv: ['/home/myhome/apps/thonny/bin/python3.10', '-u', '-B', '/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:17:37.767 INFO    thonny: sys.executable: /home/myhome/apps/thonny/bin/python3.10
22:17:37.767 INFO    thonny: sys.argv: ['/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:17:37.767 INFO    thonny: sys.path: ['/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp', '/home/myhome/apps/thonny/lib/python310.zip', '/home/myhome/apps/thonny/lib/python3.10', '/home/myhome/apps/thonny/lib/python3.10/lib-dynload', '/home/myhome/.local/lib/python3.10/site-packages', '/home/myhome/apps/thonny/lib/python3.10/site-packages']
22:17:37.767 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
22:17:37.767 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
22:17:37.768 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
22:17:37.768 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
22:17:37.768 INFO    thonny: TIME/MODS 0.035 before prepare (+73 modules)
22:17:37.769 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
22:17:37.769 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:17:37.769 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
22:17:37.819 INFO    thonny.plugins.micropython.bare_metal_backend: Poking
22:17:37.870 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:17:37.870 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
22:17:37.970 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:17:37.970 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
22:17:39.979 INFO    thonny.plugins.micropython.bare_metal_backend: Giving advice

```

```
myhome>:~/.config/Thonny$ cat frontend.log 
22:17:33.173 INFO    thonny: Thonny version: 4.0.2
22:17:33.173 INFO    thonny: cwd: /home/myhome/scripts
22:17:33.173 INFO    thonny: original argv: ['/home/myhome/apps/thonny/bin/python3.10', '-I', '-m', 'thonny']
22:17:33.173 INFO    thonny: sys.executable: /home/myhome/apps/thonny/bin/python3.10
22:17:33.173 INFO    thonny: sys.argv: ['/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/__main__.py']
22:17:33.173 INFO    thonny: sys.path: ['/home/myhome/apps/thonny/lib/python310.zip', '/home/myhome/apps/thonny/lib/python3.10', '/home/myhome/apps/thonny/lib/python3.10/lib-dynload', '/home/myhome/apps/thonny/lib/python3.10/site-packages']
22:17:33.173 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
[... cut ...]
22:17:37.690 INFO    thonny.running: Starting the backend: ['/home/myhome/apps/thonny/bin/python3.10', '-u', '-B', '/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"] /home/myhome/gitrepos/BB-ESP32-KIOSK/hardware
22:17:37.692 INFO    thonny.shell: BaseShellText.restart(False)

```

jouellnyc, 2023-06-26:
Also seeing this on an esp8266 using uP 1.20 -- ( esp8266-512k-20230426-v1.20.0.bin flashed just fine w/Thonnt)

```
21:48:14.607 INFO    thonny: Thonny version: 4.0.2
21:48:14.607 INFO    thonny: cwd: /home/john/scripts
21:48:14.608 INFO    thonny: original argv: ['/home/john/apps/thonny/bin/python3.10', '-u', '-B', '/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/ttyUSB1', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
21:48:14.608 INFO    thonny: sys.executable: /home/john/apps/thonny/bin/python3.10
21:48:14.608 INFO    thonny: sys.argv: ['/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/ttyUSB1', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
21:48:14.608 INFO    thonny: sys.path: ['/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp', '/home/john/apps/thonny/lib/python310.zip', '/home/john/apps/thonny/lib/python3.10', '/home/john/apps/thonny/lib/python3.10/lib-dynload', '/home/john/.local/lib/python3.10/site-packages', '/home/john/apps/thonny/lib/python3.10/site-packages']
21:48:14.608 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
21:48:14.608 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/ttyUSB1', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
21:48:14.612 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
21:48:14.612 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
21:48:14.613 INFO    thonny: TIME/MODS 0.047 before prepare (+73 modules)
21:48:14.613 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
21:48:14.613 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
21:48:14.614 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
21:48:14.664 INFO    thonny.plugins.micropython.bare_metal_backend: Poking
21:48:14.678 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'OK\r\n'
21:48:14.681 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'MPY: soft reboot\r\n'
21:48:14.707 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
21:48:14.728 INFO    thonny: TIME/MODS 0.115 got welcome 
21:48:14.728 INFO    thonny: TIME/MODS 0.000 bef preparing helpers 
21:48:14.728 INFO    thonny.plugins.micropython.mp_back: Preparing helpers
21:48:14.728 INFO    thonny: TIME/MODS 0.000 befsubcode 
21:48:14.730 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
21:48:14.742 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
21:48:14.875 INFO    thonny: TIME/MODS 0.146 affsubcode 
21:48:14.900 INFO    thonny: TIME/MODS 0.025 affforw 
21:48:14.900 INFO    thonny: TIME/MODS 0.000 prepared helpers 
21:48:14.900 INFO    thonny: TIME/MODS 0.000 befsubcode 
21:48:14.908 INFO    thonny: TIME/MODS 0.008 affsubcode 
21:48:14.934 INFO    thonny: TIME/MODS 0.026 affforw 
21:48:14.934 ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 151, in __init__
    self._prepare_after_soft_reboot(clean)
[... cut ...]
  File "/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 571, in _evaluate
    raise ManagementError("Script produced errors", script, out, err)
thonny.plugins.micropython.mp_back.ManagementError: Script produced errors
21:48:14.935 ERROR   thonny.plugins.micropython.mp_back: ManagementError details:
SCRIPT: __thonny_helper.print_mgmt_value(__thonny_helper.getcwd())

STDOUT: 

STDERR: Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 42, in getcwd
AttributeError: no such attribute

```

aivarannamaa, 2023-06-26:
Thank you for the report and sorry for late reply!

Could you please see whether Thonny 4.1.1 performs better?

jouellnyc, 2023-06-27:
No worries. I tried w/uP 1.19 and 1.20 w/esptool. gtkterm still works but unfortunately Thonny still unhappy with  device in the first post :

[image]

```
== BE

cat ~/.config/Thonny$ cat backend.log 
21:52:39.124 [MainThread] INFO    thonny: Thonny version: 4.1.1
21:52:39.124 [MainThread] INFO    thonny: cwd: /home/myhome/gitrepos/BB-ESP32-KIOSK/hardware
21:52:39.124 [MainThread] INFO    thonny: original argv: ['/home/myhome/apps/thonny/bin/python3.10', '-I', '-u', '-B', '/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
21:52:39.124 [MainThread] INFO    thonny: sys.executable: /home/myhome/apps/thonny/bin/python3.10
21:52:39.124 [MainThread] INFO    thonny: sys.argv: ['/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
21:52:39.124 [MainThread] INFO    thonny: sys.path: ['/home/myhome/apps/thonny/lib/python310.zip', '/home/myhome/apps/thonny/lib/python3.10', '/home/myhome/apps/thonny/lib/python3.10/lib-dynload', '/home/myhome/apps/thonny/lib/python3.10/site-packages']
21:52:39.125 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
21:52:39.125 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
21:52:39.126 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
21:52:39.126 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
21:52:39.126 [MainThread] INFO    thonny: TIME/MODS 0.038 before prepare (+73 modules)
21:52:39.126 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
21:52:39.127 [MainThread] INFO    thonny.plugins.micropython.mp_back: Sending interrupt
21:52:39.127 [MainThread] INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
21:52:39.178 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Poking
21:52:39.229 [MainThread] INFO    thonny.plugins.micropython.mp_back: Sending interrupt
21:52:39.229 [MainThread] INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
21:52:39.330 [MainThread] INFO    thonny.plugins.micropython.mp_back: Sending interrupt
21:52:39.330 [MainThread] INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
21:52:41.338 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Giving advice

==FE

~/.config/Thonny$ cat frontend.log 
21:52:24.406 [MainThread] INFO    thonny: Thonny version: 4.1.1
21:52:24.406 [MainThread] INFO    thonny: cwd: /home/myhome/scripts
21:52:24.406 [MainThread] INFO    thonny: original argv: ['/home/myhome/apps/thonny/bin/python3.10', '-I', '-m', 'thonny', '-v']
21:52:24.406 [MainThread] INFO    thonny: sys.executable: /home/myhome/apps/thonny/bin/python3.10
21:52:24.406 [MainThread] INFO    thonny: sys.argv: ['/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/__main__.py', '-v']
21:52:24.406 [MainThread] INFO    thonny: sys.path: ['/home/myhome/apps/thonny/lib/python310.zip', '/home/myhome/apps/thonny/lib/python3.10', '/home/myhome/apps/thonny/lib/python3.10/lib-dynload', '/home/myhom [... cut ...] p/esp_back.py', "{'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"] /home/myhome/gitrepos/BB-ESP32-KIOSK/hardware
21:52:39.125 [MainThread] INFO    thonny.shell: BaseShellText.restart(False)
21:52:39.157 [MainThread] INFO    thonny.plugins.micropython.mp_front: Listing serial ports

```

jouellnyc, 2023-06-27:
For the esp8266 it's the same seem there's come communication going on (seeing the soft reboot message):

[image]

 
```
== BE

ohn@wilddog:~/.config/Thonny$ cat frontend.log 
22:00:34.130 [MainThread] INFO    thonny: Thonny version: 4.1.1
22:00:34.130 [MainThread] INFO    thonny: cwd: /home/myhome/scripts
22:00:34.130 [MainThread] INFO    thonny: original argv: ['/home/myhome/apps/thonny/bin/python3.10', '-I', '-m', 'thonny']
22:00:34.130 [MainThread] INFO    thonny: sys.executable: /home/myhome/apps/thonny/bin/python3.10
22:00:34.130 [MainThread] INFO    thonny: sys.argv: ['/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/__main__.py']
22:00:34.130 [MainThread] INFO    thonny: sys.path: ['/home/myhome/apps/thonny/lib/python310.zip', '/home/myhome/apps/thonny/lib/python3.10', '/home/myhome/apps/thonny/lib/python3.10/lib-dynload', '/home/myhome/apps/thonny/lib/python3.10/site-packages']
22:00:34.130 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
22:00:34.168 [MainThread] INFO    thonny.workbench: Starting Workbench
22:00:34.353 [MainThread] INFO    thonny.workbench: Creating runner
22:00:34.356 [MainThread] INFO    thonny.workbench: Start loading plugins
22:00:34.401 [MainThread] INFO    thonny.workbench: Done loading plugins
22:00:34.515 [MainThread] INFO    thonny.workbench: Opening views
22:00:34.930 [MainThread] INFO    thonny.running: Starting the backend: ['/home/myhome/apps/thonny/bin/python3.10', '-I', '-u', '-B', '/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"] /home/myhome/gitrepos/BB-ESP32-KIOSK/hardware
22:00:35.002 [MainThread] INFO    thonny.plugins.micropython.mp_front: Listing serial ports
22:00:35.366 [Thread-1 (_listen_stdout)] INFO    thonny.running: Reader got EOF
22:00:35.475 [MainThread] INFO    thonny.shell: BaseShellText._on_backend_terminated

== BE

myhome@wilddog:~/.config/Thonny$ cat backend.log 
22:00:35.001 [MainThread] INFO    thonny: Thonny version: 4.1.1
22:00:35.001 [MainThread] INFO    thonny: cwd: /home/myhome/gitrepos/BB-ESP32-KIOSK/hardware
22:00:35.001 [MainThread] INFO    thonny: original argv: ['/home/myhome/apps/thonny/bin/python3.10', '-I', '-u', '-B', '/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:00:35.001 [MainThread] INFO    thonny: sys.executable: /home/myhome/apps/thonny/bin/python3.10
22:00:35.001 [MainThread] INFO    thonny: sys.argv: ['/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:00:35.001 [MainThread] INFO    thonny: sys.path: ['/home/myhome/apps/thonny/lib/python310.zip', '/home/myhome/apps/thonny/lib/python3.10', '/home/myhome/apps/thonny/lib/python3.10/lib-dynload', '/home/myhome/apps/thonny/lib/python3.10/site-packages']
22:00:35.001 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
22:00:35.001 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': False, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
22:00:35.005 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
22:00:35.005 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
22:00:35.005 [MainThread] INFO    thonny: TIME/MODS 0.040 before prepare (+73 modules)
22:00:35.006 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.cod [... cut ...]
  File "/home/myhome/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 566, in _evaluate
    raise ManagementError("Script produced errors", script, out, err)
thonny.plugins.micropython.mp_back.ManagementError: Script produced errors
22:00:35.299 [MainThread] ERROR   thonny.plugins.micropython.mp_back: ManagementError details:
SCRIPT: __thonny_helper.print_mgmt_value(__thonny_helper.getcwd())

STDOUT: 

STDERR: Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 42, in getcwd
AttributeError: no such attribute

```

aivarannamaa, 2023-07-04:
Can you do `import os` and `os.getcwd()` in the REPL of your device when connected with another tool?

jouellnyc, 2023-07-04:
Yes, sir, here's the output of the esp32-cam-mb:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4656
load:0x40078000,len:13284
ho 0 tail 12 room 4
load:0x40080400,len:3712
entry 0x4008064c
MicroPython v1.20.0 on 2023-04-26; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import os
>>> os.getcwd()
'/'
>>> 
```

jouellnyc, 2023-07-04:
As for the esp8266, no:

```
>>> 
>>> 
MPY: soft reboot
MicroPython v1.20.0 on 2023-04-26; ESP module (512K) with ESP8266
Type "help()" for more information.
>>> import os
>>> os.getcmd()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: no such attribute

```

jouellnyc, 2024-02-14:
Good day. 

A bit more information:

I have some other esp8266's [NodeMcu V3 CH340](https://www.aliexpress.us/item/3256805088793145.html) that I can flash ok but Thonny comes up with either the same read screen or gibberish.

gtkterm can present a repl:

```

>>> dir()
['__name__', 'os', 'machine', 'gc', 'bdev']
>>> 
MPY: soft reboot
MicroPython v1.22.1 on 2024-01-05; ESP module with ESP8266
Type "help()" for more information.

>>> import os
>>> os.getcwd() 
'/'
>>> 

```

dmliquidity, 2024-03-04:
I have a very similar issue with Thonny and I suspect I know the cause, but it's just a guess at this stage.
Some ESP32's especially the ESP32-CAM board and the common dev kit they come with use the RTS /CTS lines to reset the micro controller and put it into programming / normal mode.

In Thonny I can flash the the ESP-32 Cam fine it erases and uploads the firmware image.

I can then open up the serial monitor in VS-Code and talk the the flashed python instance.
`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4728
load:0x40078000,len:14876
ho 0 tail 12 room 4
load:0x40080400,len:3368
entry 0x400805cc
MicroPython v1.21.0 on 2023-10-05; Generic ESP32 module with ESP32
Type "help()" for more information.
>>> 2*2
4
>>> 
`

However Thonny can't and I suspect the issue is because in VS-Code I have the following serial settings.
`[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
lib_deps = nanopb/Nanopb@^0.4.7
; monitor_speed = 9600
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0`

It's the rts/dtr that matter here.
`monitor_rts = 0
monitor_dtr = 0`

I just can't figure out how to set these in thonny from the configuration.ini

I think if that this might also fix this issue [jouellnyc](https://github.com/jouellnyc) is having.

jouellnyc, 2024-05-04:
Thanks @dmliquidity I can also use the  serial  monitor in  VScode. Unfortunately  I  am pretty  much   stuck with  a bunch of  devices I can't use very well  at the moment.

rundekugel, 2025-03-27:
Hi,
same issue with 
Thonny version: 3.3.14 
6.8.0-52-generic #53~22.04.1-Ubuntu 
ESP8266 on /dev/ttyUSB0 works with picoterm, but not with Thonny
Thonny seems to set RTS/DTR not like needed.

aivarannamaa, 2025-05-04:
Thonny 5.0 will have the ability to set RTS/DTR lines