thonny/thonny #2816: Plugin installation path is not in sys.path causing websockets module to fail to load

akomakom, 2023-06-13:
## Summary

WEBREPL requires websockets module and suggests installing it in this error message:
https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/webrepl_connection.py#L27

However, installing `websockets` module via the Plug-In Manager does not resolve the error.

## Environment

* Ubuntu 22.04
* Python 3.10.6
* Thonny 4.1.1 (installed via pip)

## Workaround

`sudo pip3 install websockets` 
(Installing as user does not help)

## Analysis

1. Tools -> Open System Shell
2. Run
 ```
import sys
print(sys.path)
```
3. Output: 
```
['', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '~.local/lib/python3.10/site-packages', '/usr/local/lib/python3.10/dist-packages', '/usr/lib/python3/dist-packages']
```
4.  Tools -> Plugin Manager
5. Select `websockets`, note the value of **Installed To**: ``~/.config/Thonny/plugins/lib/python3.10/site-packages`

Note that this path is not in **sys.path**

## Links:

Original Discussion: https://forums.raspberrypi.com/viewtopic.php?t=348865 (my comment pending)

Similar issue with different cause: https://github.com/thonny/thonny/issues/2785#issuecomment-1589353989

Wind-stormger, 2023-06-28:
Useful , thanks.

aivarannamaa, 2023-07-18:
Thanks for the report and sorry for the late reply!

Thonny's backend is supposed to load optional dependencies with augmented  `sys.path`, but this wasn't properly implemented in MicroPython back-ends. 

The fix will be in 4.1.2

closed 2023-08-05 by aivarannamaa