thonny/thonny #2817: Python.exe window appears for students

KingHenryVIIISupport, 2023-06-13:
Hi,
We are running Thonny 4.0.2/Python 3.10.9/Tk 8.6.12 in an educational environment under windows 10 EDU. When a student runs Thonny, they get the Thonny window, and a python.exe console windows appears over the top. Closing this produces an error "Process ended with exit code 3221225786" - likewise, running any code via the Run menu will spawn another Python.exe window.

Students have access to run anything in program files, but running this as an admin runs Thonny and any code without the console window.

Reset the profile/settings for both test users and gone through the initial setup for both users so everything looks the same in settings etc. 

Any ideas?

aivarannamaa, 2023-06-14:
Can you give some hints for recreating such user account for reproducing the issue? Any special restrictions or policies?

KingHenryVIIISupport, 2023-06-14:
Many thanks for the reply. That would have been helpful sorry! so used to working in EDU :) 

Student accounts have redirected Desktops to a read-only folder to mandate icons, They have redirected Documents folder to a network share. Local drives are hidden (but accessible), all system folders are restricted to read only via applocker (Default rules included + Exceptions for any applications outside of the standard exe path s(program files/x86/appdata)

Not sure if I made it clear but the code inside Thonny will execute in the shell output window, but usually covered by an additional console window.

I have since found out that 4.0.2 and 4.0.1 have this but our previous version 3.3.13 doesn't and operates 'normally'

[image]

KingHenryVIIISupport, 2023-06-14:
Something else I've just noticed while looking at this is that it has moved from Program Files (x86) to Program Files (assuming an architecture change?)

aivarannamaa, 2023-07-04:
That's right -- the default variants of Thonny 4 come with 64-bit Python. There is alternative 32-bit variant available -- you could try if https://github.com/thonny/thonny/releases/download/v4.1.1/thonny-py38-4.1.1.exe works better for you.

Damian-2001, 2024-05-14:
I could solve it editing running.py and put this flag before creating subprocess in _start_background_process function:
```
        creationflags |= subprocess.CREATE_NO_WINDOW
        logger.info("Starting the backend: %s %s", cmd_line, get_workbench().get_local_cwd())
```
For thonny portable 4.1.4 (Windows)

aivarannamaa, 2024-07-12:
@Damian-2001, thanks for the tip! I'll include this flag in the upcoming version 5.0

closed 2024-07-12 by aivarannamaa