thonny/thonny #2820: version conflict with python 3.11

bcm23, 2023-06-14:
After upgrading from pytho 3.8 to 3.11 existing Thonny installation did not start;
New Thonny 4.1.1 does also not start with giving following reason:
 
pkg_resources.VersionConflict: (thonny 4.1.1 (/home/pi/.local/lib/python3.11/site-packages), Requirement.parse('thonny==3.3.10'))

aivarannamaa, 2023-06-14:
Did you try removing and reinstalling Thonny? How did you install Thonny? With `apt` or `pip`?

bcm23, 2023-06-14:
I tried removing, installing with apt , removing installing with pip3: I observed no differences in the final result

aivarannamaa, 2023-06-16:
If you're still getting error with `Requirement.parse('thonny==3.3.10'))`, then it means there are leftovers from old thonny somewhere. Perhaps it's the executable script `/usr/bin/thonny` ?

Try launching Thonny with `python3 -m thonny`.

bcm23, 2023-06-17:
you are right, the de-installation of the older thonny version went wrong.

closed 2023-06-17 by bcm23