thonny/thonny #2824: update thonnyapp

rootsmusic, 2023-06-18:
Please update [thonnyapp](https://pypi.org/project/thonnyapp/).

aivarannamaa, 2023-06-19:
Do you mean you wan't `pip install thonnyapp` to install latest `thonny` package? It should, if you don't have it installed yet. If you already have `thonny` package, then you need to do `pip install -U thonnyapp` (or `pip install -U thonny`).

rootsmusic, 2023-06-19:
@aivarannamaa  `pip install thonnyapp` installed thonny into an environment with Python 3.7.7.  Then I tried `pip install --upgrade thonny`, but it says: thonny in c:\program files\python37\thonny\lib\site-packages (3.3.14).  Should thonny update to a newer version?  Thanks.

aivarannamaa, 2023-06-19:
Thonny runs with the Python that you are using for installing it. The `thonny` package does not contain a Python interpreter.

rootsmusic, 2023-06-19:
Yes @aivarannamaa, my environment was created by Python 3.7.7.  Since Pypi has [version 4.1.1](https://pypi.org/project/thonny/), shouldn't `pip install --upgrade thonny` update it from version 3.3.14?

aivarannamaa, 2023-06-19:
> shouldn't `pip install --upgrade thonny` update it from version 3.3.14?

Yes, this should update `thonny` to latest version. If it looks otherwise then perhaps you have upgraded thonny in one environment (or for one interpreter) and tested it in another?

aivarannamaa, 2023-06-19:
Which version does `pip show thonny` show?

rootsmusic, 2023-06-19:
Package           Version
----------------- -------
astroid           2.15.5
asttokens         2.2.1
colorama          0.4.6
dill              0.3.6
docutils          0.20.1
isort             5.11.5
jedi              0.18.2
lazy-object-proxy 1.9.0
mccabe            0.7.0
mypy              1.3.0
mypy-extensions   1.0.0
parso             0.8.3
pip               23.1.2
platformdirs      3.6.0
pylint            2.17.4
pyserial          3.5
Send2Trash        1.8.2
setuptools        41.2.0
six               1.16.0
thonny            3.3.14
thonnyapp         0.9.1
tomli             2.0.1
tomlkit           0.11.8
typed-ast         1.5.4
typing_extensions 4.6.3
wrapt             1.15.0

Python 3.7.7 (I assume that thonny's version is constrained by thonnyapp's version.)

aivarannamaa, 2023-06-24:
When I create a new venv and install `thonnyapp`, then it nicely downloads `thonny==4.1.1`.

rootsmusic, 2023-06-24:
@aivaranamaa  I assume that your Python is a newer version.  I believe that thonny didn't update because my Python version is 3.7.7, which was last supported by [thonny 3.3.14](https://github.com/thonny/thonny/releases/tag/v3.3.14).  It'd be helpful, though, if pip's message is clearer than simply "Requirement already satisfied".

closed 2023-06-24 by rootsmusic