thonny/thonny #2827: Thonny ruined other debuggers for me, is there a multi language alternative of its debugger?

Ico70, 2023-06-22:
Hi after Python i am branching into Go. I tried so many ways and so many debuggers and only Thonny makes programming actually fun for me.

The way i can switch between Fast and Verbose Debugger is genius.
How was that achieved was all writen simply by @aivarannamaa pure genius or is there a library for that?

Inline variables are a middle ground, but it's so hard working in anyother debugger since i daydream about Thonny when not using Python.

If i struggled with something i would just paste the snippet in thonny. Thonny even helped me realize some mistakes when using requests becuse it was able do display that info as well.

If one was to try to build a debugger for a specific language where to start?
Is extensive knowledge of compilers and CS principles needed?
Is thonny like debugger only possible for dynamicaly typed  and interpreted languages or could it in theory work for static and compiled languages too?
I kind of whish i hadn't started using it. Other debugger pale in comparison. Is there a similar vs code extension maybe, or extension for another ide? The one that displays the value right on top  when steping through?

Thank you.

aivarannamaa, 2023-06-24:
Thank you for the kind words!

The implementation of the debuggers is done in the Thonny project itself and is not easy to extract. One may be able to reuse the general approach -- see the section "Implementation" in https://sci-hub.se/https://doi.org/10.1145/2828959.2828969

Ico70, 2023-07-02:
> Thank you for the kind words!
> 
> The implementation of the debuggers is done in the Thonny project itself and is not easy to extract. One may be able to reuse the general approach -- see the section "Implementation" in https://sci-hub.se/https://doi.org/10.1145/2828959.2828969

Thank you.
It's so weird nothing like it exists except python tutor.
Do you know of a good middleground without relying on debuggers and print debugging?

This https://marketplace.visualstudio.com/items?itemName=hediet.debug-visualizer, and the support for inline variables among a handful of IDEs and editors is the best I could find.

aivarannamaa, 2023-08-05:
> Do you know of a good middleground without relying on debuggers and print debugging?

Sorry for the late reply. No, unfortunately I can't help here.

closed 2023-08-05 by aivarannamaa