thonny/thonny #2828: macOS close unsaved File and answer no to save loses focus on Thonny

martin-guddat, 2023-06-23:
When closing an unsaved file  with File>Close (CMD W), focus does not jump to another open file

Steps to reproduce:
- Open Thonny
- Empty, unnamed file1 is open initially
- Enter some text
- Create an empty new file2, again enter some text
- Close file2 with CMD-W
- Request whether to save data before close comes up
- Select no as answer
- File is closed without save
Expected:
- Focus gets to tab and file1
- Another CMD-W would try to close file1
As-Is:
- Focus is not on any file, CMD-W does not work

Thonny 4.1.0, macOS 13.4.1

aivarannamaa, 2023-06-24:
Thanks for the report! Thonny 5.0 will have new Notebook component (#2737), and I'll deal with this issue after Notebook issue gets worked out.

aivarannamaa, 2024-03-29:
Fixed under #2737

closed 2024-03-29 by aivarannamaa