thonny/thonny #2829: Update deps for newer versions of ubuntu releases

CapeCrusader321, 2023-06-24:
Hey there @aivarannamaa  , as we had previously that Thonny is build according to dependencies of Ubuntu 18.04, this causes dependency errors in the snap package as several dependency that Thonny Depnds upon no longer exists in Ubuntu 20.04 and +, so my request is that please re-base Thonny and its dependencies to Ubuntu 20.04 and plus also as a friendly reminder Ubuntu18.04 has reached End-Of-life and is no longer supported without a commercial Ubuntu Pro license as such please consider re-basing Thonny dependencies in future release to newer Ubuntu bases like 20.04 and +
Few examples of conflicting libs include-
1) libgdbm5 (This version doesn't exist in Ubutnu 20.04 +)
2) libtinfo5 (Exists But is the legacy version and hence is redundant, libtinfo6 is recommended)
etc. 

The dependency error in snap is caused due to the fact that i build snap upon Ubuntu 22.04(core22) as building on older version significantly reduces performance of the app and i wish to use the latest versions to upkeep with the future changes.
Thanks.

aivarannamaa, 2023-06-24:
I thought snap package only takes Thonny's pure Python package as input. Are you relying on my Python+Thonny binary bundles?

CapeCrusader321, 2023-06-24:
> I thought snap package only takes Thonny's pure Python package as input. Are you relying on my Python+Thonny binary bundles?

Yes i am dumping the bundle and adjusting it according to snap confinement. It makes distribution easy.

aivarannamaa, 2023-06-26:
I'll consider moving to Ubuntu 20.04 in the upcoming Thonny 5.0. Please note that this move means Thonny+Python bundle won't work on older systems anymore.

CapeCrusader321, 2023-06-26:
> I'll consider moving to Ubuntu 20.04 in the upcoming Thonny 5.0. Please note that this move means Thonny+Python bundle won't work on older systems anymore.

Ok.
No there wont be compatibility issues, not at-least for snaps as older ubuntu or snap supporting systems  wont suffer this as they can simply download the snap package (it will automatically install core22 etc. and other needed libs are already present in the snap).
So basically snap package can overcome this and maybe also flatpak one as they are building from source i believe ?

capecrusader-121, 2023-08-06:
@aivarannamaa  Update: For the period until the new release is released and re-based i have temporarily pushed the snap to build over core18 , all thanks to the free ubuntu pro subscription. 🙃

aivarannamaa, 2024-05-30:
Starting with Thonny 5.0, I'll stop publishing Linux binaries altogether (#3210)

closed 2024-05-30 by aivarannamaa