thonny/thonny #2831: PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file'

Karagatharavishal, 2023-06-25:
**BACKGROUND:**
I am trying to connect two esp8266 together to send data to each other using UART communication, I have kept the one esp8266 as sender only and another as receiver only. below is the code for both esp8266, 

I am using the NodeMCU V3 12E ESP8266 version.

**Sender code:**
```
import machine
import time

uart = machine.UART(0, 115200, tx=machine.Pin(1, machine.Pin.OUT))  # UART number, baud rate, and TX pin

def send_data(data):
    uart.write(data)

def main():
    while True:
        send_data("Hello World!\n")  # Send "Hello World" over UART
        time.sleep(1)  # Wait for 1 second before sending the next message

main()
```
**Receiver code**
```
import machine

uart = machine.UART(0, 115200, rx=machine.Pin(3, machine.Pin.IN))  # UART number, baud rate, and RX pin

def main():
    while True:
        if uart.any():
            received_data = uart.read()
            print(received_data)
main()

```

**Error:**

When i am trying to execute this program, it is getting stuck and after some time shows the below error:
```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (thonny.plugins.micropython.mp_back.ProtocolError: Could not get raw-paste confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

**data of backend.log as below:**
[backend.log](https://github.com/thonny/thonny/files/11859213/backend.log)

aivarannamaa, 2023-06-26:
Thank you for the report!

Which Thonny version are you using? Please try the latest 4.1.1, if you are not already using it.

Karagatharavishal, 2023-06-26:
Dear Sir, Thank you for your quick response.

At present, I am using Thonny 4.1.1 and I have also tried the Thonny 4.1.0 version as well but getting the same error.

Thank you. 

Update:

I have tried to update the configuration file which is as below, but when I am restarting the thonny it goes back to the original one.

```
[general]
configuration_creation_timestamp = 2023-06-26T10:29:38.986003
language = en_US
environment = []
single_instance = False
event_logging = False
disable_notification_sound = False
debug_mode = False
ui_mode = regular
scaling = default
font_scaling_mode = default

[view]
full_screen = False
maximize_view = False
ui_theme = Windows
shellview.visible = True
filesview.visible = True
files_split = 80
assistantview.visible = False
astview.visible = False
stackview.visible = False
exceptionview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = False
outlineview.visible = False
todoview.visible = False
variablesview.visible = False
show_program_arguments = False
show_plotter = False
name_highlighting = True
locals_highlighting = False
paren_highlighting = True
syntax_coloring = True
highlight_tabs = True
highlight_current_line = False
show_line_numbers = True
recommended_line_length = 0

[LocalCPython]
last_configurations = [{'run.backend_name': 'LocalCPython', 'LocalCPython.executable': 'C:\\Program Files (x86)\\Thonny\\python.exe'}]

[ESP8266]
port = COM11
interrupt_on_connect = True
submit_mode = raw
sync_time = True
local_rtc = True
restart_interpreter_before_run = True

[run]
backend_name = ESP8266
program_arguments = 
dock_user_windows = False
pgzero_mode = False
allow_running_unnamed_programs = True
auto_cd = True
warn_module_shadowing = True
birdseye_port = 7777
run_in_terminal_python_repl = False
run_in_terminal_keep_open = True

[assistance]
disabled_checks = []
open_assistant_on_errors = True
open_assistant_on_warnings = False
use_pylint = True
use_mypy = True

[serial]
last_backend_per_vid_pid = {(6790, 29987): 'ESP8266'}

[layout]
zoomed = True
notebook_nw_visible_view = FilesView
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = None
notebook_e_visible_view = None
notebook_se_visible_view = None
west_pw_width = 195
east_pw_width = 195
nw_nb_height = 195
sw_nb_height = 195
s_nb_height = 195
se_nb_height = 195
ne_nb_height = 195

[file]
current_file = MicroPython device :: /main.py
open_files = ['MicroPython device :: /main.py']
reopen_all_files = False
make_saved_shebang_scripts_executable = True

[edit]
automatic_calltips = False
automatic_completions = False
automatic_completion_details = True
tab_request_completions_in_editors = False
tab_request_completions_in_shell = True
indent_with_tabs = False

[debugger]
frames_in_separate_windows = True
automatic_stack_view = True
allow_stepping_into_libraries = False
preferred_debugger = nicer

[shell]
clear_for_new_process = True
tty_mode = True
max_lines = 1000
squeeze_threshold = 1000
auto_inspect_values = True
```

Karagatharavishal, 2023-06-26:
Update: I have tried to send data from RP2 to nodemcu esp8266 but still facing the same problem and getting below error when i am executing the program.

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (thonny.plugins.micropython.mp_back.ProtocolError: Could not read command confirmation).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```
[backend.log](https://github.com/thonny/thonny/files/11869409/backend.log)

Karagatharavishal, 2023-06-26:
Update:

I have tried to assign the Tx pin of ESP8266 to receive the data from the RPi Pico, and it worked, But work for the Team of Thonny and Micropython is not over by this temporary solution. 

I strongly recommend both teams work together to resolve the issue for the goodwill of the engineering society.

Thanks 

Updated code is as below:

RP2 code:
```
import machine
import time

tx_pin = machine.Pin(4, machine.Pin.OUT)  # GPIO4

# Set up the UART communication
uart = machine.UART(1, baudrate=115200, tx=tx_pin)

counter = 1

while True:
    data = "hello world " + str(counter) + "\n"
    uart.write(data)
    print(data)
    time.sleep(1)
    counter += 1
```
ESP8266 code:
```
import machine
import time

rx = machine.Pin(1, machine.Pin.IN)

uart = machine.UART(0, baudrate=115200, rx=rx)

while True:
    if uart.any():
        data = uart.read()
        print(data)
        time.sleep(1)
    else:
        print("Not Receiving")
        time.sleep(1)
```

aivarannamaa, 2023-07-04:
> when I am restarting the thonny it goes back to the original one.

Thonny always rewrites configuration.ini on close. You need to edit the file while Thonny is closed.

rotellini9809, 2024-07-18:
Having the Same problem here, trying to communicate via uart with a lora antenna but the program get stuck and when I stop the execution and rerun the program I get the same error