thonny/thonny #2832: v4 very slow in school network environment

stephanmartin, 2023-06-25:
Since v4, Thonny takes age to run scripts saved in home network.
Downgrading to v3.11, slow-downs dissapear.

aivarannamaa, 2023-06-26:
Is is also slow when you re-run an already saved and not changed script? How slow is it? Do you have Files pane open?

stephanmartin, 2023-06-29:
Yes.
For example, it takes 5 seconds to run this script with a = 1000

def test(a):
    for i in range(a):
        a -= 1
    return a

With v3.3.13, it's immediate

stephanmartin, 2023-06-29:
F5 + call test(1000) in shell > 5 seconds
next call test(1000) in shell > immediate

stephanmartin, 2023-06-29:
found the reason : it's a Kaspersky issue that occurs with thonny v4
As I have to remove Kaspersky from the computers, the problem is solved for me

aivarannamaa, 2023-06-29:
Good to know!