thonny/thonny #2834: Cursor jumps to Shell after opening parentheses

dahenson54, 2023-06-27:
Every time, no matter where I am in my code editor, if I have to use the open parentheses key "(" for any reason, the cursor will immediately jump down to the Shell window, requiring me to stop what I am typing and click where I was working in the editor so I can continue what I was doing. I tried closing the Shell, but it opens it back up when I press "(".

aivarannamaa, 2023-07-02:
Thank you for the report!

This is really strange. Which OS (and version) and which Thonny version are you using? Do you use some application that remaps keys or does some other similar tricks?

dahenson54, 2023-07-02:
I was able to duplicate it consistently. It will only occur while the program is still running. If I stop the program first, I can prevent it from jumping. I am on MacOS Ventura 13.4.1, and I am using Thonny 4.0.2. I have no applications that remap keys or anything of that sort.

aivarannamaa, 2023-09-27:
I now could reproduce it when the program was running AND waiting for the user input. For this case the fix will be in Thonny 4.1.3.

Please let me know if your case was different.

closed 2023-09-27 by aivarannamaa