thonny/thonny #2835: No connection after flashing

Aluhut, 2023-06-27:
Hi there, I’m using Thonny on Mac OS Ventura 13.4.1, MacBook Pro 2020. 
After successful flashing of an esp 32/8266, it won’t be found. 
I think I read all documentation and tried everything possible. I think it’s a bug or something that I don’t know yet. 

Can someone help me please?

best

Andreas

aivarannamaa, 2023-07-04:
Thank you for the report!

What do you mean by "it won't be found"?

Aluhut, 2023-07-04:
Well pretty much as expectable... it's not available, seeable, reachable in Thonny. I don't know whats not clear so Im struggling finding a better word.

aivarannamaa, 2023-07-06:
Do you mean the device's port is not listed when you click on the lower-right corner of Thonny's window?

Or do you mean the port is not listed in "Tools => Options => Interpreter => Micropython (ESP...) => Port" ?

Or do you mean you don't get the MicroPython REPL after connecting?

Aluhut, 2023-07-08:
I’m sorry for the false alert (I’m pretty new in microcontroller), it had to do with drivers. Now it’s working fine. 
now I have another issue but I guess for that I should open another thread.

closed 2023-07-08 by Aluhut