thonny/thonny #2836: Thonny becomes unresponsive after unplugging the device

HonestQiao, 2023-06-28:
OS: macOS Big Sur 11.7.8
Device: MicroPython v1.20.0-698-g3f25a765d on 2023-06-28; ESP32S3 Box Lite (spiram octal) with ESP32S3
Thonny: 4.1.1

Test step:
1. connect the device to the usb port
2. choose the device in Thonny
3. run the follow code:
```
print("runing")
while True:
    pass
```
4. unplugging the device, error info in the shell area:
[image]
Thonny becomes unresponsive, and can't do anything until it hangs.

HonestQiao, 2023-06-29:
[image]

aivarannamaa, 2023-07-02:
Thank you for the report! I'll try to fix this for the next release.

janaldeg, 2023-07-28:
@aivarannamaa: great that you found the bug. Looking forward to the fix it in 4.1.2. Thanks. 
For the mean time with version 4.1.1, and micropython running on a Raspberry Pi PICO, I found a way to deal with it: 
In main.py, after the "import xxx"-lines, add a 10 seconds pause (sleep(10)). Then, right after connecting the device to the usb port, wait for about 5 seconds and then push Thonny's red STOP/Restart-button. One attempt is enough. Thonny will stop main.py and will be able to communicate with the device again.

tenapier, 2023-10-13:
Am using version 4.1.3 and this problem of main.py on the pico W, and "Unable to connect to /dev/cu.usbmodem14601: [Errno 2] could not open port, is still a problem for me. Any help, much appreciated!